realthunder/FreeCAD #4: OpenMPI header 'mpi.h' not detected if in /usr/include

Salamandar, 2018-09-21:
On archlinux, the OpenMPI header is `/usr/include/mpi.h`. 

`pkg-config` filters `-I/usr/include` from cflags (because it's a standard include path), so the cflags detected from `ompi-cxx` do NOT contain this token…

The problem is that the header detection code in CMake is : 

    find_file(MpidotH mpi.h PATHS ${OPENMPI_INCLUDE_DIRS} NO_DEFAULT_PATH)

The NO_DEFAULT_PATH prevents CMake from searching in `/usr/include`. The header is thus not found.
You just have to remove `NO_DEFAULT_PATH` and everything works as intended.

ceremcem, 2018-09-21:
Wouldn't it be a nice PR?

Salamandar, 2018-09-21:
You are everywhere. :p
Yeah, will do.

ceremcem, 2018-09-21:
I ammmmmmmmmm watchinggggggggg :P

Salamandar, 2018-09-21:
Haha :p
PR waiting !

realthunder, 2018-09-22:
I think this is better handled upstream. That mpi.h is introduced in b76519333f191890ccb8da168f15a959e26fabba. Maybe you can directly ask the author @ysangkok as to why `NO_DEFAULT_PATH` is necessary.

Salamandar, 2018-09-23:
> The line is there to prevent a incomprehensible linker error, since the compiler won't complain if it can't find the header AFAIK.

It's the compiler who will complain, not the linker ;) The linker does not have any knowledge of what a header is.

realthunder, 2018-09-29:
@Salamandar Have you submitted a PR to upstream for this? I'll merge your patch here.

luzpaz, 2020-10-30:
@Salamandar ping

Salamandar, 2020-12-14:
Hi, sorry for the disappearing. It looks like this issue is not anymore : searching for OpenMPI is done by `pkg_search_module`. Closing.

luzpaz, 2020-12-14:
@Salamandar ticket is still open

Salamandar, 2020-12-14:
Not anymore :)

closed 2020-12-14 by Salamandar