realthunder/FreeCAD #8: A new spreadsheet behaviour?

yrabbit, 2018-10-20:
Hello
LinkDev branch, commit 9f80debbc0fdaec4d100538ed781eb3e37df71b3

I can no longer just type text in the cell. For some reason, the “=” sign is added automagically, which leads to an error.
Maybe I should somehow enter the text differently?
[image]

ceremcem, 2018-10-20:
I'm not using LinkDev branch, but would you try `'This is text`?

realthunder, 2018-10-20:
> I'm not using LinkDev branch, but would you try `'This is text`?

Yes, a starting `'` can force plain text. 

Please be warned. I am doing a major refactor of expression inside LinkDev branch. You may encounter many strange behavior with spreadsheet before it become matured. The file you saved in LinkDev may not be able to open once the branch is merged.

yrabbit, 2018-10-20:
Thanks, ```'``` does the trick :)

closed 2018-10-20 by yrabbit

ceremcem, 2018-10-20:
> Please be warned. I am doing a major refactor of expression inside LinkDev branch.

We're all holding tight :D