realthunder/FreeCAD #9: ‘gp_Pln pln’ has incomplete type

J-Dunn, 2018-11-30:
Hi, 

I have coin4 and FreeCAD-pivy installed.  I now got 36% into building  FreeCAD-LinkStage3 and hit this error. 

```
/svn/FreeCAD-LinkStage3/src/Mod/Part/App/TopoShapePyImp.cpp: In member function ‘PyObject* Part::TopoShapePy::findPlane(PyObject*)’:
/svn/FreeCAD-LinkStage3/src/Mod/Part/App/TopoShapePyImp.cpp:2126:16: error: aggregate ‘gp_Pln pln’ has incomplete type and cannot be defined
         gp_Pln pln;
                ^~~
make[2]: *** [src/Mod/Part/App/CMakeFiles/Part.dir/build.make:1549: src/Mod/Part/App/CMakeFiles/Part.dir/TopoShapePyImp.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1856: src/Mod/Part/App/CMakeFiles/Part.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```
Any suggestions how to takle the problem?

thx

realthunder, 2018-11-30:
I just made a commit to add the header. However, the fact that you have this error probably means that you are not using OCCT 7 (the CAD kernel), but OCE instead. If not sure, please attach your `<build_dir>/CMakeCache.txt` file here and let me check. It is strongly recommended to update OCCT to at least version 7. Official FreeCAD binary distribution is using OCCT 7.3 now.

J-Dunn, 2018-11-30:
```
//disable usage of occt-config files
OCCT_CMAKE_FALLBACK:BOOL=OFF

//The directory containing a CMake configuration file for OCE.
OCE_DIR:PATH=/lib64/oce-0.18
```
There is only one occurance of "OCCT" in that file. There are tons of "OCT". 

I do recall seeing something about OCT VS OCCT, but don't recall where. How do I fix that?

realthunder, 2018-11-30:
Looks like there is no official build for OCCT 7 on Fedora, again. I found a [post](https://forum.freecadweb.org/viewtopic.php?t=11994&start=50) regarding Fedora build. Maybe you can ask around. It is not just OCCT, but other companion libraries as well.

I'll modify my branch to compile with OCE. Meanwhile, if you really want, you can try download OCCT source from [here](https://www.opencascade.com/content/latest-release), and compile it, with `cmake -DCMAKE_BUILD_TYPE=Release` and `make`. But I image you'll get into trouble along the way. 

In order for FreeCAD to use OCCT, you need to add `-DFREECAD_USE_OCC_VARIANT="Official Version"` when running `cmake`.

J-Dunn, 2018-12-01:
Thanks, I'd already figured we had to go to OCCT for the source. I tried your defines:

`cmake -DFREECAD_USE_OCC_VARIANT="Official Version" -DCMAKE_BUILD_TYPE=Release  ..`

It seemed OK but reported:

> CMake Warning:
>   Manually-specified variables were not used by the project:
> 
>     FREECAD_USE_OCC_VARIANT

Looks like it may matter ;)

realthunder, 2018-12-01:
FREE_USE_OCC_VARIANT is for FreeCAD cmake, not OCCT. Once you get OCCT built, you need to re-run cmake in your FreeCAD build directory, with this `-DFREECAD_USE_OCC_VARIANT="Official Version"`

J-Dunn, 2018-12-01:
BTW , anon download is possible (preferable). 
https://www.opencascade.com/content/packaging-again-debian
https://git.dev.opencascade.org/gitweb/?p=occt.git;a=snapshot;h=refs/tags/V7_3_0;sf=tgz

realthunder, 2018-12-01:
I have fix build using OCE. In case you can't get through with OCCT, you can fallback to OCE.

J-Dunn, 2018-12-01:
Thanks, 

I got the latest version just after you posted it. I have built and installed OCCT , fine. 

```
-- Checking for module 'Coin'
--   Package 'Coin', required by 'virtual:world', not found
-- Found Spnav: /usr/lib64/libspnav.so  
-- Using default python: -python2.7
-- libshiboken built for Release
-- Found PySide Tools: /bin/pyside-uic, /bin/pyside-rcc
-- -- matplotlib-2.2.3 has been found.
-- Platform is 64-bit, set -D_OCC64
-- Build type: 
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
bzr: ERROR: Not a branch: "/svn/FreeCAD-LinkStage3/".
svn: E155007: '/svn/FreeCAD-LinkStage3' is not a working copy
Unknown version control
/svn/FreeCAD-LinkStage3/linkstage3_build/src/Build/Version.h.out written
```
Coin4 is installed. Am I right in thinking that this does not affect the build process? Hopefully it will be found at run time. 

The git error is because I unpacked from a zip download. It seems this is less "fatal" than indicated since the make process is running.

realthunder, 2018-12-01:
Looks okay. And yes, this is a fork. You can probably keep the upstream master around for comparison in case you found something wrong. There are quite a lot changes. Oh, and don't forget to [build](https://github.com/realthunder/solvespace#building-on-linux) python solvespace.

J-Dunn, 2018-12-01:
thanks, I have upstream master at /usr/local/bin/FreeCAD. Presumably your version will install to the same place, so I need to 'make uninstall ' the original ( just keeping the build directory ). 

If I follow you install instructions correctly I just need to copy tree to Ext/freecad/  and rename it asm3. No build process. 

since your slvs instructions rely on git , I'd better get a git copy of asm3, rather than my zip version, since I don't think the git commands will work from the zip contents.

realthunder, 2018-12-01:
You don't need to uninstall the upstream. You can run my branch directly from its build location. And git clone asm3 into your `<build_dir>/Ext/freecad` directory. I have prebuilt binary inside asm3 repo, which may or may not work depending on your system library version. You can try it first.

realthunder, 2018-12-01:
Inside asm3 directory, run

```
git submodule update --init --recursive
```

Then cd to `slvs` sub directory, and run

```
mkdir build
cd build
cmake -DBUILD_PYTHON=1  ..
make
```
Detailed instruction can be found [here](https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction#solvespace). After done, copy out the binary as described in the instruction.

J-Dunn, 2018-12-01:
Many thanks. I've just done _slvs  and the main LinkStage3 and just completed without tripping up ( about 4h worth ). 

Those were the instructions I was following, so there is nothing to be done inside asm3 apart from 
`make _smvs`

I should be good to go .... at last.

J-Dunn, 2018-12-01:
OK , running from the build dir, as normal use it seems to work and opens up a block I have made with older versions. Now to see whether I can assemble something !!

Many thanks for you help in building all this. It would have been impossibly long without some guidance.

J-Dunn, 2018-12-01:
I am trying to follow this explanatory guide:
https://github.com/realthunder/FreeCAD_assembly3/wiki/Create-Skeleton-Sketch
but having added assembly container; added a default workplane, selected it I don't find the indicated icon for adding sketch plane. Neither does there seem to be anything on the Assembly3 menu which matches.
> 
Select the new work plane, and click AddSketchPlane to add a sketchPlane constraint

Are these instructions in sync with the current build? 

thx

realthunder, 2018-12-01:
Click the icon with three dots to reveal extra buttons.

J-Dunn, 2018-12-02:
Thanks, this thing has more icons that Photoshop !  It's hard to find your way around at first. 

I'm trying to do the cylinder+cube asm3 example but I can't get rid of the move part tool. In the anim, you just seem to click somewhere and it disappears but it's not clear what you do. This is problem I'm having persistently with FreeCAD. There must be some way to get out of a tool and back to the arrow cursor that I'm missing.  Can you help?

thx

luzpaz, 2020-12-14:
@J-Dunn do you mind updating what the status of this ticket is?

J-Dunn, 2020-12-14:
Sorry, I have not used asm3 in about 18mo.  I found great in what it did but it got unusably slow once the assembly got a few levels deep. I even bought an SSD since this was meant to speed it up but it didn't help.  asm3 served as a great intro to FreeCAD but I admit I can't even remember where I was with this issue. It is possible I had to fall back to using AppImage files instead of a full dev build.

realthunder, 2020-12-15:
@J-Dunn Please make a new issue for the problem. And it will be better if you can provide a file showing the problem.

closed 2020-12-15 by realthunder