realthunder/FreeCAD #11: Sketch constraints and binary ops

yrabbit, 2019-04-25:
Constraints on expressions stopped working, and I just can't figure out how to fix it. In the formulas, I refer to the constraints of the same sketch, and the program does not like it.
And so not only with division, but also with subtraction.
This is a pretty big project that broke, but I can try to do something small to repeat the situation if necessary.

Branch: LinkStage3, Commit:2b73b3033d50f03fd07fae9d3c86850c67f4028e

[https://peertube.social/videos/watch/42a0b56a-366d-4ccd-8a2e-735d2c10fb38](Here is a video illustrating the situation.)

[image]
[image]

realthunder, 2019-04-25:
You are using a very recent revision! This is kind of risky. There are a few big commits for expression recently. I should have made those changes in my private branch. Anyway, I'll try to fix it ASAP.

realthunder, 2019-04-25:
BTW, in case you are in a hurry. You can try revert to 0312da8bda22d4cba4e646c12d0f804a692f6efa.

realthunder, 2019-04-25:
Should be fixed. Please try.

closed 2019-04-25 by realthunder

yrabbit, 2019-04-25:
Last commit (58a3b25b0dc584e7631da3e64eb7dba54bc7f1b3) does not go through compilation.
Maybe something with my build system, I'll try revert to 0312da8.
---
src/Gui/Tree.cpp:977:62: error: 'GetLinkRecursive' is not a member of 'App'
     for(auto link: App::GetApplication().getLinksTo(obj,App::GetLinkRecursive|App::GetLinkArray))
---

realthunder, 2019-04-25:
Missed some local changes. Added now. Please try again.

yrabbit, 2019-04-25:
Thanks. Expressions work now.

commit 58a3b25b (2019-04-25): Sketcher: fix constraint expression

Fixes #11
commit f848f7ed (2022-07-02): Add cups interface for printing

See https://snapcraft.io/docs/cups-interface

Closes #11