realthunder/FreeCAD #12: Copying of spreadsheets

yrabbit, 2019-04-26:
Copying spreadsheets leads to not quite understandable errors for me.

Branch: LinkStage3 Commit: d03d5d66e1ddec6ea4aef88065906754cc788f87

[https://peertube.social/videos/watch/7315d6e4-d863-4292-8762-c0dcadf2d9de](Video)

[image]
[image]

realthunder, 2019-04-26:
Please sync with my branch and try again. If there is still problem, please prepare a small file to demo the problem.

yrabbit, 2019-04-26:
After sync, everything works as it should. Thanks.

closed 2019-04-26 by yrabbit