realthunder/FreeCAD #20: opencamlib

tiket18, 2019-09-17:
ArchLinux
 pacman -Qs opencam
local/opencamlib-git 2019.07-1
    Multi-Purpose CNC Toolpath Library

OpenCamLib is not working!
OpenCamLib is not working!
Path workbench activated

apparently it makes sense to provide either the creation of the necessary links during compilation and installation, or detection

realthunder, 2019-09-18:
I have not use OCL before. Did you compile FreeCAD yourself? Maybe you can find some clue [here](https://www.freecadweb.org/wiki/OpenCamLib)

tiket18, 2019-09-18:
Yes, I compiled it myself, or rather, my laptop. The process was based on the link https://aur.archlinux.org/packages/freecad/ PKGBUILD and changed the link to the source from the github there, set https://github.com/realthunder/FreeCAD (This is a stable working solution). He assembled and installed all the dependencies before assembly.
OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18765 (Git)
Build type: Release
Branch: LinkStage3
Hash: 89c07f94b57b27269327f4122ca06073326ef19b
Python version: 3.7.4
Qt version: 5.13.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Russian/Russia (ru_RU)

realthunder, 2019-09-18:
I didn't see opencamlib in the PKGBUILD you linked, which is expected, as it is only a runtime dependency. Did you compiled opencamlib yourself too? Have you check the FC wiki about opencamlib I linked above? Maybe yours is not compiled with python3.

tiket18, 2019-10-14:
Yes, the change in PKGBUILD is

    cmake -DBUILD_PY_LIB = ON -DUSE_PY_3 = ON ../src -Wno-dev

resolved the issue of error. However, now another pops up and let's try to bring it to the discussion by email? w@wau.spb.ru

closed 2020-05-22 by tiket18