realthunder/FreeCAD #23: Where to get SMESH?

akloster, 2019-10-28:
When following the official build instructions with this fork, I get errors about SMESH, and SMESH_EXTERNAL is off by default. I guess I have to set it to ON and find SMESH somewhere?

```
[ 31%] Built target SMESH
Makefile:129: recipe for target 'all' failed
```

I haven't found a debian package for that, nor do I know which of the projects called SMESH is required and where to install it.

Any idea?

realthunder, 2019-10-28:
The message shows SMESH is built already. The error must be somewhere else. Can you please post the entire CMake output?

akloster, 2019-10-28:
Here is the CMake Output: http://dpaste.com/1R7T3SA

akloster, 2019-10-28:
Make output, from the point where the trouble seems to start: https://pastebin.com/T5qQ52b9

realthunder, 2019-10-28:
According to CMake output, SMESH does seem to have some problem. Maybe you can explicitly disable external search using `-DFREECAD_USE_EXTERNAL_SMESH=OFF`.

Also, you python executable setting ( `-DPYTHON_EXECUTABLE=/usr/bin/python3.7`) is wrong. It should point to an executable file instead of a directory.

akloster, 2019-11-05:
My current cmake command:
```
cmake -DBUILD_QT5=ON -DPYTHON_EXECUTABLE=/usr/bin/python3.7 -DPYTHON_INCLUDE_DIR=/usr/include/python3.7/ -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.7m.so    -DFREECAD_USE_EXTERNAL_SMESH=OFF
```
I noticed this in the cmake output:
```
-- Python:              [/usr/bin/python3.7] [-python2.7.x86_64-linux-gnu]
```

And now I am getting an error about shiboken2 wanting to compile something for 2.7 and missing symbols when linking:

```
/usr/lib/x86_64-linux-gnu/libshiboken2-python2.7.x86_64-linux-gnu.so: Warnung: undefinierter Verweis auf »PyString_FromFormatV«
/usr/lib/x86_64-linux-gnu/libshiboken2-python2.7.x86_64-linux-gnu.so: Warnung: undefinierter Verweis auf »PyBuffer_FromReadWriteMemory«
/usr/lib/x86_64-linux-gnu/libshiboken2-python2.7.x86_64-linux-gnu.so: Warnung: undefinierter Verweis auf »PyString_Type«
collect2: error: ld returned 1 exit status
src/Main/CMakeFiles/FreeCADMain.dir/build.make:123: recipe for target 'bin/FreeCAD' failed
make[2]: *** [bin/FreeCAD] Error 1
CMakeFiles/Makefile2:866: recipe for target 'src/Main/CMakeFiles/FreeCADMain.dir/all' failed
```

realthunder, 2019-11-06:
Is this `/usr/bin/python3.7` a directory or an executable on your computer? Your last cmake output suggests this is a directory? If so, you should correct it to point to the actually python interpreter executable. I think shiboken2 is only available for python 3.

akloster, 2019-11-06:
/usr/bin/python3.7 is a Python executable from the standard repository.

I don't quite understand why cmake still wants to link against Python 2.7.

realthunder, 2019-11-06:
Oops, sorry, I was meant to say `/usr/lib/python3.7`. And I see that you latest cmake command has already corrected this. Have you performed a clean cmake run? Maybe you can ask around in the forum. I remember there was quite a fuss about finding the correct shiboken2 lib.

akloster, 2019-11-08:
I managed to compile both master and the realthunder fork using Python 3.6.

luzpaz, 2022-01-18:
@akloster please close ticket

closed 2022-01-19 by luzpaz