realthunder/FreeCAD #24: error install,  This, apparently, the consequences of the update gcc

tiket18, 2019-11-08:
The program does not compile.
I am simply compiling https://github.com/FreeCAD/FreeCAD.git = it is going to be installed, it works.

I collect https://github.com/realthunder/FreeCAD.git the assembly crashes with an error -
`
../src/3rdParty/salomesmesh/src/SMESH/SMESH_MeshEditor.cpp:4159:25: warning: “u” may be used without initialization in this function [-Wmaybe-uninitialized]
 4159 | newUV.SetX (ElCLib :: InPeriod (newUV.X (), u1, u2));
      | | | ~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
==> ERROR: A build () failure has occurred.
    Interrupt...
==> ERROR: Makepkg was unable to compile freecad-linkstage3-git.
==> Restart assembly freecad-linkstage3-git? yes / no [y / n]
==> ----------------------------------------------- -----------
`

This, apparently, the consequences of the update gcc -

`pacman -Qs gcc
local/gcc 9.2.0-4 (base-devel)
    The GNU Compiler Collection - C and C++ frontends
local/gcc-fortran 9.2.0-4
    Fortran front-end for GCC
local/gcc-libs 9.2.0-4 (base)
    Runtime libraries shipped by GCC
local/lib32-gcc-libs 9.2.0-4 (multilib-devel)
    32-bit runtime libraries shipped by GCC`

But the fact is that the program of the main branch is assembled and installed, but this branch is not.

realthunder, 2019-11-08:
Can you please post the whole ninjia output? The quoted output below is only a warning,

> ../src/3rdParty/salomesmesh/src/SMESH/SMESH_MeshEditor.cpp:4159:25: warning:

tiket18, 2019-11-15:
everything seems to be science -
git clone https:/...
cd FreeCad
cmake -GNinja      -DCMAKE_BUILD_TYPE="Release"    -DCMAKE_INSTALL_PREFIX='/usr/lib/freecad'     -DCMAKE_INSTALL_LIBDIR='/usr/lib/freecad/lib'     -DCMAKE_INSTALL_DOCDIR='/usr/share/freecad/doc'     -DCMAKE_INSTALL_DATADIR='/usr/share/freecad'     -DFREECAD_USE_OCC_VARIANT="Official Version"     -DFREECAD_USE_EXTERNAL_PIVY=ON    -DFREECAD_USE_QT_FILEDIALOG=ON     -DBUILD_QT5=ON     -DPYTHON_EXECUTABLE=/usr/bin/python3.8 .

-- Qt5WebKitWidgets:    5.212.0
-- Shiboken2:           5.13.2 [/usr/lib64/cmake/Shiboken2-5.13.2] []
-- PySide2:             5.13.2 []
--  IncludeDir:         Unable to find, python version mismatch?
-- PySide2Tools:        [/usr/bin/pyside2-uic] [/usr/bin/pyside2-rcc]
-- Freetype:            2.10.1
-- OpenGLU:             /usr/lib/libGLU.so [/usr/lib/libGLU.so][/usr/include]
-- Coin3D:              [Coin] []
-- SPNAV:               [/usr/lib/libspnav.so] [/usr/include]
-- Matplotlib:          3.1.1
-- Rift:                not enabled (BUILD_VR)
-- Doxygen:             not found
=================================================
Now run 'cmake --build FreeCAD' to build FreeCAD
=================================================

-- Configuring done
-- Generating done
-- Build files have been written to: FreeCAD

ninja -j4
ninja: error: dependency cycle: src/Mod/TechDraw/Gui/mtextedit.h -> src/Mod/TechDraw/Gui/mtextedit.h

tiket18, 2019-11-23:
/FreeCAD/src/Base/swigpyrun.inl:75:31: error: incorrect use of incomplete "PyInterpreterState" type {aka "struct _is"}
   75 | PyObject *modules = interp->modules;
      | ^~
In file included from /usr/include/python3.8/genobject.h:11,
                 from /usr/include/python3.8/Python.h:121,
                 from /home/m/FreeCAD/src/Base/PyExport.h:42,
                 from /home/m/FreeCAD/src/Base/swigpyrun.cpp:25:
/usr/include/python3.8/pystate.h:20:8: note: forward declaration "PyInterpreterState" {aka "struct _is"}
   20 | struct _is;
      | ^~~
make[2]: *** [src/Base/CMakeFiles/FreeCADBase.dir/build.make:729: src/Base/CMakeFiles/FreeCADBase.dir/swigpyrun.cpp.o]
make[2]: *** Waiting for tasks to be completed...
make[1]: *** [CMakeFiles/Makefile2:315: src/Base/CMakeFiles/FreeCADBase.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

realthunder, 2019-11-24:
I have cherry-picked the fix from [upstream](https://www.freecadweb.org/tracker/plugin.php?page=Source/view&id=14470). Please sync my branch and try again.

tiket18, 2019-11-24:
git clone https://github.com/realthunder/FreeCAD.git
cd FreeCAD/
make .
cmake --build /home/us/FreeCAD -j8
...
[ 12%] Copying /home/us/FreeCAD/src/Mod/Draft/TestDraft.py to /home/us/FreeCAD/Mod/Draft/TestDraft.py
[ 12%] Copying /home/us/FreeCAD/src/Mod/PartDesign/fcgear/involute.py to /home/us/FreeCAD/Mod/PartDesign/fcgear/involute.py
[ 12%] Generating Draft_rc.py
[ 12%] Copying /home/us/FreeCAD/src/Mod/PartDesign/fcgear/svggear.py to /home/us/FreeCAD/Mod/PartDesign/fcgear/svggear.py
[ 12%] Copying /home/us/FreeCAD/src/Mod/Draft/Resources/icons/DraftWorkbench.svg to /home/us/FreeCAD/data/Mod/Draft/Resources/icons/DraftWorkbench.svg
[ 12%] Built target RobotScripts
[ 12%] Built target PartDesignScripts
[ 12%] Copying /home/us/FreeCAD/src/Mod/Draft/Draft_rc.py to /home/us/FreeCAD/Mod/Draft/Draft_rc.py
[ 12%] Built target Draft
make: *** [Makefile:130: all] error 2

or

cmake . -DPYSIDE_LIBRARY="/usr/lib/libpyside2.cpython-38-x86_64-linux-gnu.so" -DBUILD_QT5=ON -DPYTHON_EXECUTABLE=/usr/bin/python3  -DOPENMPI_INCLUDE_DIRS=/usr/include -DPYSIDE_INCLUDE_DIR="/usr/include/PySide2" -DPYSIDE_PYTHONPATH="/usr/lib/python3.8/site-packages/PySide2" -DPYSIDE_TYPESYSTEMS="/usr/share/PySide2/typesystems" -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX="/usr/lib/freecad" -DCMAKE_INSTALL_DATAROOTDIR="/usr/share" -DCMAKE_INSTALL_DOCDIR="/usr/share/freecad/doc" -DFREECAD_USE_OCC_VARIANT="Official Version" -DFREECAD_USE_EXTERNAL_PIVY=ON  -DCMAKE_INSTALL_LIBDIR=/usr/lib/freecad/lib -DPYTHON_INCLUDE_DIR=/usr/include/python3.8 -DPYTHON_PACKAGES_PATH=/usr/lib/python3.8/site-packages/

cmake --build /home/us/FreeCAD -j8
...

[ 22%] Copying /home/us/FreeCAD/src/Mod/Fem/femobjects/_FemSolverCalculix.py to /home/us/FreeCAD/Mod/Fem/femobjects/_FemSolverCalculix.py
[ 22%] Copying /home/us/FreeCAD/src/Mod/Fem/femobjects/FemConstraint.py to /home/us/FreeCAD/Mod/Fem/femobjects/FemConstraint.py
[ 22%] Built target FemScriptsTarget
[ 22%] Copying /home/us/FreeCAD/src/Mod/Arch/Arch_rc.py to /home/us/FreeCAD/Mod/Arch/Arch_rc.py
[ 22%] Built target Arch
make: *** [Makefile:130: all] error 2

tiket18, 2019-11-24:
if many times in a row, despite mistakes, to give the command of assembly, up to periodically falling down to 53%

FreeCAD/src/Mod/Part/App/PropertyTopoShape.cpp:399:18: error: "os" was not declared in this scope; did you mean "Cos"?
  399 | OSD_OpenStream(os, File, ios::out);
      | ^~
      | Cos
/home/us/FreeCAD/src/Mod/Part/App/PropertyTopoShape.cpp:399:28: error: "ios" was not declared
  399 | OSD_OpenStream(os, File, ios::out);
      | ^~~
make[2]: *** [src/Mod/Part/App/CMakeFiles/Part.dir/build.make:1060: src/Mod/Part/App/CMakeFiles/Part.dir/PropertyTopoShape.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1726: src/Mod/Part/App/CMakeFiles/Part.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

realthunder, 2019-11-24:
Can you please provide your system details, like what Linux variant are you using, which version, etc.

tiket18, 2019-11-25:
At the same time, the usual freecad build from git is successfully built, launched and running. Here is the information about it -

OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18776 (Git)
Build type: Release
Branch: master
Hash: 5cbf18b8bfce394742d08e5115867e14be02be1d
Python version: 3.8.0
Qt version: 5.13.2
Coin version: 4.0.0a
OCC version: 7.4.0
Locale: Russian/Russia (ru_RU)

tiket18, 2019-12-04:
possibly
https://bugreports.qt.io/browse/PYSIDE-939

realthunder, 2019-12-05:
I'll do a merge with upstream this month. We'll see how it goes then.

tiket18, 2019-12-07:
a good solution would probably be to collect as an appimage

tiket18, 2020-05-22:
The installation question is closed - all the work is done.

closed 2020-05-22 by tiket18