realthunder/FreeCAD #25: LinkStage3 branch lacks "Export selected geometries as ..." Sketcher actions

bgamari, 2019-11-10:
The [Assembly3 0.10.2 AppImage](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.10.2) has a very handy pair of actions in Sketcher: "Export selected geometries as separate child objects" and "Export selected geometries as single child objects".

However, I am finding that my own build of the `LinkStage3` branch (specifically commit f97f6b41b172b5ac65b95b62b5bc43087f44e539) and `asm3` branch (specifically https://github.com/realthunder/FreeCAD_assembly3/commit/b7da4ecc416f759dd44c6ffef22225dd76d035bd) seems to lack these actions. Were these introduced in another branch? What exactly are the AppImages built from?

realthunder, 2019-11-11:
Where do you get the commit f97f6b41b172b5ac65b95b62b5bc43087f44e539. I could find it.

bgamari, 2019-11-11:
Whoops; my apologies. That was the `solvespace` commit. I am using FreeCAD commit 2ca84062e736ccc563fad584e675df4804c57031.

realthunder, 2019-11-11:
The buttons are still there. You can't find them on toolbar?

[image]

bgamari, 2019-11-12:
Indeed they are not in the toolbar with 2ca8406. Which commit introduced this feature?

realthunder, 2019-11-12:
Can you please post a screenshot of the toolbar when editing a sketch? The buttons were added by [this](https://github.com/realthunder/FreeCAD/blob/2ca84062e736ccc563fad584e675df4804c57031/src/Mod/Sketcher/Gui/Workbench.cpp#L208) line.

bgamari, 2019-11-12:
Done.

[image]

realthunder, 2019-11-12:
That's strange. Are you sure you are running my branch? Can you paste what's in the 'About' window?

Also when you scroll down the task panel here, did you see the Element listbox headers?

[image]

bgamari, 2019-11-12:
Oh dear; indeed I was building a different tree than I thought. My apologies for the hassle.

closed 2019-11-12 by bgamari