realthunder/FreeCAD #27: LinkDev: linear pattern transformation bug

dulouie, 2020-02-29:
Hi realthunder, i love your recent partdesign changes in LinkDev. Partdesign is the most important workbench for me. While playing around with patterns i found a transformation bug in linear pattern. Please have a look at the attached file, the second flower has a wrong orientation.
[linearpattern_transfail.zip](https://github.com/realthunder/FreeCAD/files/4270519/linearpattern_transfail.zip)

realthunder, 2020-03-01:
Fixed. Thanks for report. You'll need to manually recompute the transformed feature.

dulouie, 2020-03-03:
I can comfirm, it's fixed.
Btw: What are your plans with multiple solids in partdesign?  And what's about your enhanced external references in sketcher? Is there a possibility to see both in 0.19?

realthunder, 2020-03-03:
I'll add property multiple body support into PartDesign::Body in next release. You'll be able to create multiple linear history inside the body.

I have done some preliminary preparation for this in LinkDev. There is the new Split feature in PartDesign to let you turn on/off individual solids. It is similar to Part WB Split. You can create a split with no input to split all solids, or add a datum plane to split at a chosen location, or change its 'Mode' to 'CompSolid' to split one solid into multiple. The split solids will be grouped under the Split feature. By default, the first solid will be enabled. Double click each solid to turn on/off.

Another improvement is to allow SubShapeBinder or Link to be used as a base for a new body, or used inside boolean operation on other body. So you can already do some multi-body design, just not so convenient, because each body live in its own coordinate space. You'll have to use binder to import any external geometry references.

The sketcher enhancement won't be merged in 0.19. It will be added to the next upstream release, and so is the topo naming stuff.

commit d5575243 (2020-03-01): PartDesign: fix feature transformed

Fixes #27
commit a8e2a7de (2022-07-02): Add numba dependency for fcFEM development

Closes #27
closed 2020-03-17 by realthunder