realthunder/FreeCAD #28: LinkDev: can't compile windows

rgbertoni, 2020-03-03:
Hi realthunder, thank you for your amazing work. Your implementations allowed me to start my project.
I´m trying to compile LinkDev on windows but I can't success. I'm getting the following error:
=>>> freecad\src\base\Base64.h(73): error C2059: syntax error: '__declspec(dllexport)'
I'm using windows 10 and FreeCADLibs_12.1.4_x64_VC15. Same error with FreeCADLibs_12.1.2_x64_VC15.

realthunder, 2020-03-03:
Made some changes. Please try.

rgbertoni, 2020-03-04:
Hi the Base64.h was solved but now I got another ones.

Error	C2512	'std::basic_istream<char,std::char_traits<char>>': no appropriate default constructor available	FreeCADBase	D:\AtConcept\rgbertoni\FreeCAD\src\Base\Reader.cpp	641	

[error-compile.log](https://github.com/realthunder/FreeCAD/files/4287655/error-compile.log)

realthunder, 2020-03-05:
Should be working now. I have tried it on my side.

realthunder, 2020-03-05:
Oops, no. Still has problem. Hang on there.

realthunder, 2020-03-05:
Okay, now it should be fine, I hope...

rgbertoni, 2020-03-05:
It worked !!! Thank you.

closed 2020-03-05 by rgbertoni