realthunder/FreeCAD #32: toponaming issue PartDesignExample

dulouie, 2020-05-12:
Hey, I tested your toponaming algorithm a little harder today. I'm really pleased with the results. So I can use your branch for my important stuff right now. Just stupid construction with sketches on surfaces :)

But I noticed a bug.
1. open the PartDesignExample file from the FreeCAD start screen
2. open the Sketch of the Pad
3. draw an additional circle in the middle of the tower
4. unfortunately, the moat (Pocket) has now jumped to the bottom

I have worked out a workaround:
First put the tip on the pad and draw the circle. Then set the
the tip back to pocket002.
The moat has stayed in the right place this time.

realthunder, 2020-05-13:
This is not really a bug per se. It is caused by the fact that you are opening a legacy document, hence no topo name there yet. Those names are only generated after recomputation. Therefore, editing the model before any recomputation may give you the same problem as in upstream. I just added a commit to prompt the user for recomputation upon opening a legacy document.

dulouie, 2020-05-13:
All right, at first I thought that old files didn't work, but then I read the "auto-generated element name" messages and was impressed. So maybe old corrupted files can be repaired?

dulouie, 2020-05-13:
I just compiled your last commit and the dialog is still hanging.

realthunder, 2020-05-13:
Should be fixed now. I have also changed the 'auto-generated element name' warning message to 'Pending recompute...', because the name is not really generated on restore. It is only that the object is marked for recompute.

dulouie, 2020-05-13:
It works perfectly. Meanwhile, I use your branch for my big stuff. If I find any faults, I report them immediately. Besides, I'm already totally excited about the upcoming multibody properties.

closed 2020-05-14 by dulouie