realthunder/FreeCAD #33: Does not display the limitations set in the thumbnail (verticality, perpendicularity, tangent, etc.).

tiket18, 2020-05-22:
Does not display the limitations set in the thumbnail (verticality, perpendicularity, tangent, etc.).

OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23166 (Git)
Build type: Release
Branch: makepkg
Hash: d033670da9b24e5876ad49c1d5e917e4b7ca9687
Python version: 3.8.2
Qt version: 5.14.2
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Russian/Russia (ru_RU)

At the same time the main branch of the appimage does not have this problem.

And the other question is: how far behind the main branch is the Path layout in this branch? - There are active works in this branch and I would like to understand how and when new products from the main path branch get into this branch.

realthunder, 2020-05-27:
I'll do a merge with the upstream for my next release update.

tiket18, 2020-05-27:
if only for Path, it is not necessary (the layout is updated frequently) -
https://forum.freecadweb.org/viewtopic.php?f=15&t=46532#p401546

tiket18, 2020-06-04:
the restriction display disappears from the screen after setting a new Symmetry restriction (S key)

realthunder, 2020-06-04:
Yes, this problem is just fixed. I'll make a new release later today.

luzpaz, 2022-01-18:
@realthunder this ticket can be closed

tiket18, 2022-01-19:
Yes. The problem persists. In upsrim, too. In a constraint-rich sketch, quite often when you set the symmetry constraint and turn on the "automatically remove redundancies" option, FC turns on the "symmetry" requirement and does not remove the "vertical" or "horizontal" that becomes redundant.
That's not all :)
It also likes to cause a crash (undecidable, etc.) in a saturated sketch when trying to set a "point match" - when trying to close multiple arcs in a point. The workaround is to "put the point on the line".