realthunder/FreeCAD #34: GUI - Axial is reset

tiket18, 2020-05-27:
when editing the PartDisign position... any (almost) changes (offset X, U, rotation around the axes - i.e. any, but we do NOT change the Axial value, where the default is 0mm) when you click OK or Apply, an error message is displayed - the existing Axial value is completely erased and you have to manually go into this field, enter 0mm and click Apply again.

tiket18, 2020-05-27:
The reason, I think, is that when you open the position editing window, all fields will have default values with the dimension (mm), and Axial will have only values, without mm.

realthunder, 2020-05-29:
Can you please post a screenshot, or better, capture a screencast showing the problem?

tiket18, 2020-05-29:
[image]

realthunder, 2020-05-31:
Should be fixed. I'll make a release later today. Thanks for reporting.

commit 3d3d4dac (2020-05-31): Gui: fix QuantitySpinBox

Fixes #34
closed 2020-05-31 by realthunder