realthunder/FreeCAD #35: does not start after the update. It's not going to.

tiket18, 2020-06-03:
~$ freecad
freecad: error while loading shared libraries: libpyside2.cpython-38-x86_64-linux-gnu.so.5.14: cannot open shared object file: No such file or directory

find / -name libpyside2.cpython-38-x86_64-linux-gnu.so*

/usr/lib/libpyside2.cpython-38-x86_64-linux-gnu.so
/usr/lib/libpyside2.cpython-38-x86_64-linux-gnu.so.5.15
/usr/lib/libpyside2.cpython-38-x86_64-linux-gnu.so.5.15.0

I don't understand why I need to specify the library program with the build number, for example libpyside2.cpython-38-x86_64-linux-gnu.so.5.15.0.
if there is a universal sl?

 /usr/lib/libpyside2.cpython-38-x86_64-linux-gnu.so

tiket18, 2020-06-03:
extra/python-shiboken2 5.15.0-1 [installed]
extra/shiboken2 5.15.0-1 [installed]

tiket18, 2020-06-04:
../src/Mod/Image/Gui/ImageView.cpp:481:37: предупреждение: «int QWheelEvent::delta() const» устарело: Use angleDelta() [-Wdeprecated-declarations]
  481 |        int numTicks = cEvent->delta() / 120;
      |                                     ^
In file included from /usr/include/qt/QtGui/QMouseEvent:1,
                 from ../src/Mod/Image/Gui/ImageView.cpp:23:
/usr/include/qt/QtGui/qevent.h:219:16: замечание: объявлено здесь
  219 |     inline int delta() const  { return qt4D; }
      |                ^~~~~
[2923/3828] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/Workbench.cpp.o
ninja: build stopped: subcommand failed.

realthunder, 2020-06-04:
> freecad: error while loading shared libraries: libpyside2.cpython-38-x86_64-linux-gnu.so.5.14: cannot open shared object file: No such file or directory

Did you get the error when running your own build or my pre-built image?

> ../src/Mod/Image/Gui/ImageView.cpp:481:37: предупреждение: «int QWheelEvent::delta() const» устарело: Use angleDelta() [-Wdeprecated-declarations]

This is a warning. The error must be somewhere else. Please attach the complete build log.

tiket18, 2020-06-04:
When you build it from source.
glad to hear that there's a version of appimage where you can download it?

tiket18, 2020-06-04:
Problems appeared after updating my shiboken laptop to version 5.15. I made the launch of the earlier copied by cheating the system with two symbolic links. But the new compilation doesn't work.

realthunder, 2020-06-04:
I remember see some forum post about the shiboken2 troubles. It is kind of tricky I guess. You can search the forum for that. I haven't encountered similar problem, though.

You can download my pre-built at https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11

tiket18, 2020-06-04:
Very good, thank you so much for your work.

realthunder, 2020-06-04:
You're welcome. It is built from my FreeCAD branch LinkStage3, and bundled with assembly3 workbench.

tiket18, 2020-06-14:
In the main branch, there have been many updates in recent days related to qt5.15 & Py2, maybe they would help to fix the build?

tiket18, 2020-06-26:
please update the appimage (it's still from June 18th) and if possible update the Path workbench from the main version - I have completely switched to your build.

closed 2020-10-20 by tiket18