realthunder/FreeCAD #36: Export of linked parts is not always working

lilaL, 2020-06-17:
Exporting a part with linked objects is not always working well.
[image]

ceremcem, 2020-06-17:
Hi, 

FYI: @realthunder would probably require the example FreeCAD project, steps to reproduce this bug and the version of the FreeCAD. If you provide those right away, it's possible to save some time.

realthunder, 2020-06-17:
OBJ/STL export does not support App::Link at the moment. As a work around. You can use `Part -> Copy -> Make simple copy` to copy the top App::Part object, and then export that copy.

luzpaz, 2022-01-18:
@realthunder trying to sort through some of these back logged tickets. Any reason to keep this open?