realthunder/FreeCAD #37: Expression

rgbertoni, 2020-07-17:
Hi, I'm facing an error when I try to set a string to an object property expression:

GuiCommand: Line
----------------
GuiCommand: Line
Pick first point
Pick next point
Illegal storage access...
<Exception> Illegal storage access! Please save your work under a new file name and restart the application!
in property binding 'Unnamed#Line.Label'
0 <App> Document.cpp(4035): Failed to recompute Unnamed#Line: Illegal storage access! Please save your work under a new file name and restart the application!
in property binding 'Unnamed#Line.Label'
Traceback (most recent call last):
  File "D:\AtConcept\compile\install\Mod\asm3\freecad\asm3\mover.py", line 488, in slotRecomputedDocument
    Assembly.resumeSchedule()
  File "D:\AtConcept\compile\install\Mod\asm3\freecad\asm3\assembly.py", line 3640, in resumeSchedule
    cls.schedule()
  File "D:\AtConcept\compile\install\Mod\asm3\freecad\asm3\assembly.py", line 3626, in schedule
    if not cls._ScheduleTimer.isSingleShot():
<class 'TypeError'>: bad argument type for built-in operation
Recompute failed! Please check report view.

closed 2020-07-18 by realthunder

realthunder, 2020-07-18:
Thanks for reporting. Image updated.

rgbertoni, 2020-07-18:
Greate, thank you.

commit 889c1e21 (2020-07-18): App: fix pyObjectToAny() string conversion

Fixes #37
commit 21fa2a55 (2020-07-18): App: fix pyObjectToAny() string conversion

Fixes #37