realthunder/FreeCAD #39: Box Selection

rgbertoni, 2020-08-05:
Hi,

It is possible to make box selection via python? I mean, a function passing the screen coordinates and the freecad do selection inside the rectangle formed by the coordinates.
I searched in the forum, but as I'm using your branch a guess you could help me.
Anyway, thanks for your good work.

Regards.

closed 2020-08-07 by realthunder

realthunder, 2020-08-07:
I have just exposed this functionality to Python. It will be available in next release. Sample usage:

```python
>>> Gui.ActiveDocument.ActiveView.Viewer.getPickedList([(200,200),(300,300)])
[(<Part::PartFeature>, 'Face1'), (<Part::PartFeature>, 'Face4')]
```

It can also do single point high level picking, e.g.

```python
>>> Gui.ActiveDocument.ActiveView.Viewer.getPickedList((764,255))
[(<Part::PartFeature>, 'Face1'), (<Part::PartFeature>, 'Shell1'), (<Part::PartFeature>, 'Solid1'), (<Part::PartFeature>, 'Face3')]
```

Its docstring is shown below

[image]

rgbertoni, 2020-08-07:
Wow, great, thank you. This exactly what I need.

commit f2a3a53f (2020-08-07): Gui: expose box (element) selection to Python

Via View3DInventorPy.getPickedList(), e.g.

    Gui.ActiveDocument.ActiveView.Viewer.getPickedList([(0,0),(100,100)])

Closes #39
commit 5d5e08e1 (2020-08-07): Gui: expose box (element) selection to Python

Via View3DInventorPy.getPickedList(), e.g.

    Gui.ActiveDocument.ActiveView.Viewer.getPickedList([(0,0),(100,100)])

Closes #39