realthunder/FreeCAD #41: FreeCADGui.Selection results with linked parts

abetis, 2020-09-10:
Hello @realthunder ,

I think the FreeCADGui.Selection doesn't work properly with linked parts.
Lets say I have the same part linked to the document several times with different names.

the addSelection callback is called with the following parameters:
doc - document name
obj - object name
sub - part of the object name
pnt - coordinates of the click (not interesting for that case)

I'm trying to add a feature to Assembly4 and try to use Selection callbacks.
When I click on the part link in the tree view, I get expected results:
doc - current document name
obj - the linked part name as appearing in the tree view

When I click on a body object of the the same part I get results that will not allow be to determine what part the body belongs to:
doc - name of the linked document
obj - "Part__Feature" string

I understand that the body belongs to the linked part, but without getting the name of the linked part in current document, the callback is useless in that case...

Is there a workaround for that? Or a fix will be needed?

Thanks.

realthunder, 2020-09-10:
`FreeCADGui.Selection.addObserver()` accepts a second argument or `int`. By default it is `1`, meaning the selected sub-object is auto resolved, and `sub` will only contain sub-element reference, like `Face`, `Edge`, etc. If you pass '0', then you'll get the top parent object in `obj`, and the full sub-object path in `sub`. You can use `obj.getSubObjectList(sub)` to obtain a list of objects along the path.

abetis, 2020-09-10:
Thanks. It returns more "logical" response now, but it looks like I don't understand something in the naming...
Clicking on the part body in the list view returns a name like "m4x12_allen_screw_3.Part__Feature." (with the point at the end)
Clicking on the same body in the 3D view returns a name like "m4x12_allen_screw_3.Part__Feature.Face79"

I want to use that information to get the selected object from the Gui.ActiveDocument using the getObject method.
In both cases getObject can't find anything.
If I pass it only the beginning of the string (m4x12_allen_screw_3), I get the correct object.
I don't think I should deal with the naming in the script to get the first portion... I might be missing something here.

Didn't check yet all the Selection code, but there might be "getSelected" method, or something similar that returns the selected object. Want to understand how can I handle the returned name in case I won't find the needed method

abetis, 2020-09-10:
On of the problems with the callback is that all the parameters are names of the objects and not a reference to them, so I can't call any of the methods, such as obj.getSubObjectList(sub)

realthunder, 2020-09-10:
Something like this

```python
def addSelection(self, doc, obj, sub, pos):
    App.getDocument(doc).getObject(obj).getObjectList(sub)
```

abetis, 2020-09-10:
print(App.getDocument(doc).getObject(obj).getObjectList(sub))
<class 'AttributeError'>: 'App.Part' object has no attribute 'getObjectList'

The returned 'doc' is the active document, the 'obj' is the "Model" branch in Assemby4 tree, getObject returns Part, which doesn't have the getObjectList...

realthunder, 2020-09-10:
oops, sorry, should be getSubObjectList()

abetis, 2020-09-10:
Thanks!
Will the returned value always be [part, link, feature] in my case?
Or I should check some special cases?
I do check that the second returned value is App::Link type and derived from App::Part.

realthunder, 2020-09-10:
It returns the entire hierarchy of the current selected object. So if there are more nested parts, you'll get longer list.

abetis, 2020-09-10:
ok, understand.
Will check the list from end to start to find the needed link object.

abetis, 2020-09-10:
@realthunder , will appreciate your help with the followup regarding that.

What is the meaning of the "." at the end of the name?

I get to the proper object name, try to pass it to the setSelection method, that doesn't find the sub-object.
The subObject.Name is a clean and nice "m4x12_allen_screw_3"
The setSelection method expect it to be "m4x12_allen_screw_3." (with the point at the end).
I don't mind adding the point in the code, but I think something is off if I have to do it by myself...

realthunder, 2020-09-10:
Because the hierarchical selection capability is added later. Originally `sub` is used only for sub-element, like `Face` and `Edge`. It is later extended to include object path. To disambiguate object name and sub-element name, we demand that each sub-object name in `sub` must end with a `.`

abetis, 2020-09-10:
Ok, thanks for the answers, I was able to implement the feature I wanted with your help.

closed 2020-09-10 by abetis