realthunder/FreeCAD #42: "Send To Python Console" from tree view sends wrong information for some elements of the linked part

abetis, 2020-09-11:
Don't know if it's related to links and can't explain why the feature works for most of the entries, but not for "LCS_Origin" and "Variables"...

I use Assembly4 file where main document name is "roller", which have linked part named "m4x12_allen_screw_1".
The "roller" has child objects:
Parts - empty group
Model - part object that has all the linked objects in that entry

Inside the "Model" there are:
Origin - coordinate
LCS_Origin - coordinate
Constraints - group
Variables
m4x12_allen_screw_1 - link to m4x12_allen_screw document

inside the "m4x12_allen_screw_1" there is the same structure as in the "Model"...

If I right click and select "Send To Python Console" option on different items in the tree, I would expect to see matching code appearing in the python code, which is ok for all the entries, but not for the m4x12_allen_screw_1::LCS_Origin for some reason.

Right click on Model produce:
- "obj = App.getDocument("roller").getObject("Model")" --> good
Right click on Model::Variables produce:
- "obj = App.getDocument("roller").getObject("Variables")" --> good
Right click on Model::LCS_Origin produce:
- "obj = App.getDocument("roller").getObject("LCS_Origin")" --> good
Right click on Model::m4x12_allen_screw_1 produce:
- "obj = App.getDocument("roller").getObject("m4x12_allen_screw_1")" --> good
Right click on m4x12_allen_screw_1::PartBody (imported STEP) produce:
- "obj = App.getDocument("m4x12_allen_screw").getObject("Part__Feature")" --> good
Right click on **m4x12_allen_screw_1**::Variables produce:
- "obj = App.getDocument("**roller**").getObject("Variables")" --> bad
Right click on **m4x12_allen_screw_1**::LCS_Origin produce:
- "obj = App.getDocument("**roller**").getObject("LCS_Origin")" --> bad

OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22292 (Git)
Build type: Release
Branch: master
Hash: f09d48d0fe00590997b88b52b610210ac606951c
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

realthunder, 2020-09-12:
Can you post your testing file?

abetis, 2020-09-12:
I've prepared some files to separate from a larger project.
The ZIP include the needed files, open the test.FCStd, it links to the shaft and spacer.
[test.zip](https://github.com/realthunder/FreeCAD/files/5212291/test.zip)

realthunder, 2020-09-12:
This bug was already fixed in my branch. But you are using the upstream FreeCAD. I have just submitted a [PR](https://github.com/FreeCAD/FreeCAD/pull/3863) there.

luzpaz, 2022-01-14:
@realthunder this ticket can be closed

luzpaz, 2022-01-18:
@abetis care to close this ticket?

abetis, 2022-01-18:
Unfortunately I don't have an environment to check the fix.
Probably close. At worst we'll re-open it.

Thanks.

luzpaz, 2022-01-18:
@abetis please close it, i have no permissions to :disappointed:

closed 2022-01-18 by abetis