realthunder/FreeCAD #45: Windows Specific - UI Bug

obelisk79, 2020-09-26:
Unable to dock panel views on any side other than the left side of the screen. They can be undocked from the left, redocked, including nesting multiple layers of docks. I'm unsure if this is related to the libpack. I compiled with FreeCADLibs_12.1.6_x64_VC15.

Build was clean off latest commit and gave no errors. Also, panel position and setup is not persistent between restarts/workbench changes. I will try to provide some screenshots to illustrate better what is happening.

obelisk79, 2020-09-26:
For reference, I was attempting to follow the setup directions from this video:
https://www.youtube.com/watch?v=s7O0gIvXnLU&t=326s

obelisk79, 2020-09-26:
Before Swtiching workbenches:
[(https://ibb.co/2ZYt4M3)]

After Switching workbenches:
[https://ibb.co/1vxSVYs]

realthunder, 2020-09-26:
I am in the middle to making a new release. If you want an early taste, please compile and test the `LinkMerge` branch. There is new drag and drop implemented for dock widgets. The default branch `LinkStage3` is current in the middle of implementing that feature. `LinkMerge` branch has many extra fixes. You can drag the title bar of a dock widget and drop it in any side with overlay or not. You can also drag the side of a title bar to resize the overlay.

obelisk79, 2020-09-26:
Thank you for your work and the response. I will compile and test.

obelisk79, 2020-09-26:
I built the LinkMerge branch and still have the same behavior. This does appear to be unique to windows, the panels behave correctly when built for linux.

realthunder, 2020-09-26:
Hmm... it works on my computer. Can you post your About information here? Also can you capture some screencast of try to dock on right side by dragging?

Can you also try my last window [release](https://github.com/realthunder/FreeCAD_assembly3/releases) and see if it works?

obelisk79, 2020-09-26:
https://ibb.co/Jn89p51

Here you can see the docking behavior and persistence issue
I'll check the release and report back.

Here is the about info from my build: 
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25085 (Git)
Build type: Release
Branch: LinkMerge
Hash: 660af0699e1750e9cdd40bded4c7a0725a36b8ba
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

obelisk79, 2020-09-26:
Ok, so I loaded up your release build and the docking behavior is all correct. So I suppose I must have an issue in my build-chain.

realthunder, 2020-09-26:
Can you please sync the [LinkMerge](https://github.com/realthunder/FreeCAD/commits/LinkMerge) branch first? You didn't get today's commits. Maybe some github syncing issue. However, today's commit only fixes the workbench switching problem. I didn't add any code for the drag and drop. I did use LibPack 12.1.2 for my release build. But that shouldn't matter as both version uses the same Qt 5.12.1.

Can you try this? Drag one dock widget out and drop it in the middle so that it is floating. Now hold the `Shift` key and then start dragging the title bar. This shall bypass my custom drag and drop logic and let Qt handle the dragging. See if you can dock to the right.

Also, how about docking to the top? Is it working?

obelisk79, 2020-09-26:
Ok, I haven't rebuilt to the latest commit just yet, however, I noticed that loading up my previous build, it had saved all of the docking and UI positioning I made while using your build. Additionally I could now undock and re-dock on the right side.
As you requested, I also undocked to floating, and then holding shift, clicked and docked a panel onto the top. Docking to the top ONLY worked when holding shift however. Docking to the right now oddly works both ways. Although the UI docking animation and behavior seems better when bypassing your drag and drop.

obelisk79, 2020-09-27:
Update, 
Built using your latest commits to the LinkMerge branch as of 9am EDT and noticing the same behavior. I will try to build as 32bit and also x64 using LibPack 12.1.2 to see if there is any difference in behavior with those.

realthunder, 2020-09-27:
Can you please try resize the main window and then see if can dock or not?

obelisk79, 2020-09-27:
If I resize it still will not dock, unless I bypass with shift. 

I have noticed that if I dock a panel, restart FreeCAD I am then able to dock/undock to that side without problems. If a side had no panels docked at start, then I am unable.

realthunder, 2020-09-27:
The referenced commit above should fixed the problem. Please sync `LinkMerge` and test.

obelisk79, 2020-09-27:
Now I cannot dock any panel anywhere unless I bypass with shift.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25097 (Git)
Build type: Release
Branch: LinkMerge
Hash: 6b485169d279249fc692221725571a9f24266cb4
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

realthunder, 2020-09-27:
Really? Hmm... it's getting weirder... I can actually reproduce your problem based on your last description, on both Linux and Window, i.e. having no widget docked on right side, restart FreeCAD, and it is not able to dock to right. But with that patch, the problem is gone on both my computers. Can you please capture a screencast again?

obelisk79, 2020-09-27:
Here you go: https://ibb.co/wcDZjBh

realthunder, 2020-09-28:
Can you please keep the state as what's like before you start the screencast. Close FreeCAD, and send me your configuration file. Should be at `C:\Users\<your_user_name>\AppData\Roaming\FreeCAD\system.cfg` and `data.cfg`.

obelisk79, 2020-09-28:
[FreeCAD.zip](https://github.com/realthunder/FreeCAD/files/5289368/FreeCAD.zip)

obelisk79, 2020-09-28:
didn't have a data.cfg file

realthunder, 2020-09-28:
Should be working now... finger crossed

obelisk79, 2020-09-28:
Looks like proper behavior now,  excellent work and thank you!

commit ba4ec889 (2020-09-27): Gui: fix empty overlay docking side size initialization

Fixes #45
closed 2020-09-28 by obelisk79