realthunder/FreeCAD #46: Object.Parents returns wrong information when multiple links exist to the same document

abetis, 2020-09-26:
Hi.
Please see the attached example file:
[asm_tuto2.zip](https://github.com/realthunder/FreeCAD/files/5286948/asm_tuto2.zip)

Using Assembly4.
There are 2 "arm" links in the assembly: arm001 and arm2
Expand both arm's entries in the tree view.
Right click on **arm001**.Body002 in the tree, select "Send to python console".
In the console run "obj.Parents", the result will be:
[(<Part object>, '**arm2.Body002**.'), (<Part object>, 'Body002.')]
Right click on arm2.Body002 in the tree, select "Send to python console".
In the console run "obj.Parents", the result will be:
[(<Part object>, 'arm2.Body002.'), (<Part object>, 'Body002.')]

OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22517 (Git)
Build type: Unknown
Branch: gui-command-enhancements
Hash: 7616153b3c31ace006169cdc2fdafab484498858
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

abetis, 2020-09-26:
Is it related to #42 ?

realthunder, 2020-09-27:
I'll submit a PR to upstream later. You can manually apply the commit and try.

commit e9d348e2 (2020-09-27): App: fix recursive App::getLinksTo()

Fixes #46
closed 2020-09-28 by realthunder