realthunder/FreeCAD #47: Link's Shape property

abetis, 2020-09-27:
Hi.

I see that link object has a hidden Shape property.
It doesn't appear in the object's properties when you type "obj.", but if I access it, I can get it's values.
Is it supposed to be that way?

It could be a great feature to have a shape of linked object, but I saw that the returned values of link.Shape are wrong :(

realthunder, 2020-09-27:
Can you please post a file showing the wrong shape?

abetis, 2020-09-28:
Hmm... I'm trying to find the file where I saw the problem, but can't reproduce it.
I've used many old tutorial files, which probably were created with some old or custom build versions.
Will keep an eye on that and report if I'll see it again.

Anyway, the "Shape" property is not listed for Link object in the console, while it is accessible.

And something I want to confirm:
Does the link attachment offsets are applied to the Shape information as well?
I mean, if an object has offset of 100mm on X axis, will the Center of BoundBox have that offset as well?

abetis, 2020-09-28:
Here... Found it...
[test.zip](https://github.com/realthunder/FreeCAD/files/5291195/test.zip)
Open the test1.FCStd file, it have a link to goTUBE 336mm.FCStd
The goTUBE 336mm.FCStd file was created with imported STEP.

Send "goTUBE 336mm.FCStd" entry from the tree to Python console
```
>>> obj.Shape.BoundBox
BoundBox (-15.8585, -1e+100, -15.975, 15.8585, 1e+100, 15.975)
```

Send "goTUBE 336mm.FCStd" entry from the tree to Python console
```
>>> obj.Shape.BoundBox
BoundBox (-15.8585, 0, -15.975, 15.8585, 336, 15.975)
```

realthunder, 2020-09-29:
What did your second `obj.Shape.Boundbox` refer to? I think you made a typo. Are you referring to the tube model? The Link to model gives the compound of all its children shape, including those datum lines.

> Anyway, the "Shape" property is not listed for Link object in the console, while it is accessible.

It is not listed because it is not a real FreeCAD property. It is actually implemented as a Python attribute. It is dynamically calculated by calling `Part.getShape()`. In other words, `getattr(link, 'Shape')` is equivalent to `Part.getShape(link)`. And yes, it will adjust the returned shape's placement according to the link placement.

The same things goes for `App.Part`, i.e. `getattr(part, 'Shape')` is equivalent to `Part.getShape(part)`, which returns a compound of its children in part's placement.

abetis, 2020-10-02:
> What did your second `obj.Shape.Boundbox` refer to? I think you made a typo. Are you referring to the tube model? The Link to model gives the compound of all its children shape, including those datum lines.

Sorry, copy&paste didn't work when typing...
I meant goTUBE_336mm link and "4103-0032-0336" body of the object.

So, based on your answer, Shape is not relevant for Assembly4 since its using FC's elements that define an object such as line... That might be a problem on the long run if Assembly4 will become popular...

> It is not listed because it is not a real FreeCAD property. It is actually implemented as a Python attribute. It is dynamically calculated by calling `Part.getShape()`. In other words, [...]

I'm not familiar with the code that much, but I as saw, Visibility is also an attribute that run show/hide functions based on the call, but it's still shown in the python console.