realthunder/FreeCAD #48: FreeCAD fails to start on built of LinkStage3 #921df5da81

Cyclonit, 2020-09-28:
Hi,

I am trying to build and run FreeCAD using the LinkStage3 branch (mock info):

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 
Build type: ?
Branch: LinkStage3
Hash: 921df5da8119a331861655e0142985be978fe927
Python version: 3.6.8
Qt version: 5.12.1.0
Coin version: 4.0.0
OCC version: ?
Locale: English/US (en_US)
```

The build finishes without any errors. Upon trying to run FreeCAD_d.exe directly, I get a few seconds of loading indicator on the cursor but FreeCAD does not start. When I try running the exe from the command line, I get the following error message:

```
$ ./FreeCAD_d.exe
Fatal Python error: Py_Initialize: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'
Current thread 0x000035f0 (most recent call first):
```

My `FreeCADLibs_12.1.6_x64_VC15\bin` is in my PATH variable and no other python version is installed. No errors occur during configuration through Cmake.

Any ideas on what might be causing this issue?

obelisk79, 2020-09-28:
I'm not sure what your experience level with compiling software is, however I found this video really made the process straightforward. I've not had issues building following his steps in conjunction with the information on the Wiki

https://www.youtube.com/watch?v=s4pHvlDOSZQ&feature=youtu.be

https://wiki.freecadweb.org/Compile_on_Windows

realthunder, 2020-09-29:
The error either means you have more than one Python installation, or the python module cannot be found. It is recommended to copy the contents of libpack's bin directory into your build output directory.

Cyclonit, 2020-09-29:
I was able to complete a build just now. As I pointed out before I don't have any other Python installations on this system. I am unsure as to what the exact issue was, but it was most likely related to copying the libpack's bin directory to the proper place. I followed the video @obelisk79 posted to the letter and it works. Thanks for both of your help.

closed 2020-09-29 by Cyclonit