realthunder/FreeCAD #49: Cannot switch to Sketcher Workbench

Cyclonit, 2020-09-29:
Hi,

after having built your fork of FreeCAD successfully, I cannot open most workbenches in FreeCAD. Building "ALL_BUILD" as debug finished without any errors. When I try to open the Sketcher workbench, the following error message appears:
[image]

My configuration according to FreeCAD
```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git)
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Germany (en_DE)
```

Upon running the tests I get this error:
```
$ ./FreeCAD_d.exe --run-test 0
test_import_draft (drafttests.test_import.DraftImport)
Import the Draft module. ... ok
test_import_draft_geomutils (drafttests.test_import.DraftImport)
Import Draft geometrical utilities. ... ok
test_import_draft_svg (drafttests.test_import.DraftImport)
Import Draft SVG utilities. ... ok
test_import_draft_vecutils (drafttests.test_import.DraftImport)
Import Draft vector utilities. ... ok
test_arc (drafttests.test_creation.DraftCreation)
Create a circular arc. ... Illegal storage access...
```

I am using the Python version from the Libs:
```Python 3.6.8 (v3.6.8:3c6b436a57, Mar 26 2019, 15:42:50) [MSC v.1916 64 bit (AMD64)]```

[FreeCAD.log](https://github.com/realthunder/FreeCAD/files/5299754/FreeCAD.log)
[crash.log](https://github.com/realthunder/FreeCAD/files/5299752/crash.log)
[crash.dmp.zip](https://github.com/realthunder/FreeCAD/files/5299765/crash.dmp.zip)

obelisk79, 2020-09-29:
Did you setup cmake to copy all of the libpack files to your build/install dir? It appears FreeCAD is looking for a library that isn't present in its sub directories.

Cyclonit, 2020-09-30:
Yes I have:
[image]

luzpaz, 2022-07-04:
Closing ticket

closed 2022-07-04 by luzpaz