realthunder/FreeCAD #50: Manjaro/Arch linux build failure

obelisk79, 2020-09-30:
Issue building on Manjaro/Arch linux:
main FreeCAD/FreeCAD branch builds ok
LinkStage3 fails.

Builds ok on Win10 not on linux
cmake error log attached
[CMakeError.log](https://github.com/realthunder/FreeCAD/files/5305248/CMakeError.log)

This may be a dependency related issue, although I have triple checked all dependencies appear to be installed in accordance with the guidelines set forth here: https://wiki.freecadweb.org/Compile_on_Linux

realthunder, 2020-09-30:
Can you please post the complete console output?

obelisk79, 2020-09-30:
I'm building right now, and I think I corrected the issue by adding 
#include <QPainterPath>

to /src/Gui/OverlayWidgets.h
I'll report once complete

obelisk79, 2020-09-30:
Confirmed, adding the include statement corrected the build problem. Unsure why I was able to build on Win10 but not linux without it.

obelisk79, 2020-09-30:
I believe unrelated, but I get the following in console when I run FreeCAD

`[obelisk79@obelisk79-manjaro bin]$ ./FreeCAD
FreeCAD 0.19, Libs: 0.19R25101 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

<Command> Command.cpp(1839): duplicate command Std_TreePreSelection
<Command> Command.cpp(1839): duplicate command Std_TreeSelectAllInstances
connect failed: No such file or directory
Coin info in SoGLSLShaderObject::printInfoLog(): GLSLfragmentShader  log: '0:67(1): error: no function with name 'DirectionalLight'
'
Coin error in SoGLSLShaderProgram::ensureLinking(): error: 'GL_INVALID_VALUE' (set envvar COIN_GLERROR_DEBUGGING=1 and re-run to get more information)
`

realthunder, 2020-09-30:
Not sure why you got the `duplicate command` warning, but that's not fatal.

The `GLSLfragmentShader` error is due to an older `libCoin` which means that shadow rendering won't work. If you really want the shadow, you'll have to download [Coin3D](https://github.com/realthunder/coin) source code and compile it yourself. You'll also need to compile pivy as well.

obelisk79, 2020-09-30:
Understood, thank you for the response.

closed 2020-09-30 by obelisk79