realthunder/FreeCAD #51: Part Design: Splitting by sketch

bgamari, 2020-10-01:
9ac862bb9431b42b323bf27d9d1732511d58335d introduced the very useful ability to split a Part Design feature into multiple fragments using a two-dimensional tool. However, it's quite unclear precisely what objects are expected to work as a tool:

 * I have confirmed that datum planes indeed work
 * Attempting to move any 3D geometry under a Split node results in an error (e.g. "Failed to replace object Split with Pad").
 * Sketches can be moved under a Split node and selected to be a tool but this doesn't appear to do any splitting

In particular, I was hoping to be able to use the 2D surface resulting from the non-solid "extrusion" of a sketch normal to the sketch plane. However, I have been unable to find a way to achieve this. Any hints?

realthunder, 2020-10-02:
You can create a (green) SubShapeBinder to make a face from the sketch and then use it for splitting. With the current release, there is limited way of using the splitted solids, mainly through binder from other body. You can either use the those solid as base feature to create another body. Or use the solid as tool to perform boolean operation in other body.

The soon coming next release will have a new feature called `Extrusion`, which allows to extrude from either vertex, edge or face to create non-solid for various uses. There will also be better support for manipulating multiple solids inside the same body.

bgamari, 2020-10-02:
Thank you. @realthunder!

bgamari, 2020-10-02:
By the way, is there any documentation written describing what subshape binders actually are? It seems that they are similar to shape binders but bind only a subset of geometry from the base feature. However, I can't tell whether that is the whole story.

realthunder, 2020-10-02:
See document [here](https://wiki.freecadweb.org/PartDesign_SubShapeBinder). Basically, you use this binder for cross Part/Body binding. It will track the relative placement of the bound object and the binder's container body. Note that if you move the bound object to another Part container, you'll need to double click the binder to resync it.

bgamari, 2020-10-04:
Thanks @realthunder!

closed 2020-10-04 by bgamari