realthunder/FreeCAD #52: UI element positioning bug

obelisk79, 2020-10-11:
Good Afternoon,

When I open a document, or restart FreeCAD some UI  elements are not restored to their previous positioning.

Here is what it looks like:
[image]

Here is what I repeatedly set up my view to look like:
[image]

realthunder, 2020-10-11:
I'll fix that.

BTW, combo view with model and task is probably going to be obsoleted in the future. I can't think of any real use case of two tree view, just waste of resources. If you want a split property and tree view, you can set the tree view mode in preference dialog as 'Tree view and property view'. And you'll get a stand alone property view that you can move around independent of task view. Just dock it together with the task view if you want. The inside panels of combo view are not controlled by overlay manager.

obelisk79, 2020-10-12:
Thanks for the info!

commit bb9ae8c4 (2020-10-12): Gui: save/restore ComboView splitter sizes

Fixes #52
closed 2020-10-12 by realthunder