realthunder/FreeCAD #53: Does not open (breaks) files created by itself

tiket18, 2020-10-16:
За отсутствием свежего appimage собрал из исходников, с нуля в новой сборке создал эскизы, управляющие программы. Закрыл. с утра открываю - все поломано. Открываю тот же файл через аппимадж - все на месте. Открываю через аппимадж основной ветки - условно все на месте.

For lack of fresh appimage collected from sources, from scratch in the new build created thumbnails, control programs. Closed. I open in the morning - everything is broken. I open the same file through the appimage - everything is in place. I open it through the main branch apimage - everything is in place.

realthunder, 2020-10-16:
There is new AppImage available. Please go to the release page, and look for image with name `Daily` in it. The latest release is today 20201016, and it is in sync with upstream master. If you still experience any problem, please post some sample file here.

tiket18, 2020-10-16:
В свежем appimage (как и старом) проблемы нет. Поставил пересобрку из исходников.
There is no problem with the fresh appimage (like the old one). I put a re-assembly from the sources.

tiket18, 2020-10-16:
And yes, thanks for the fresh appimage - very helpful.

realthunder, 2020-10-16:
As shown in your screen captured About page, your source is not in sync with my branch. I did fix related problem just today.

tiket18, 2020-10-16:
[image]
No, the situation has not changed in the locally built program. It was compiled without errors.

tiket18, 2020-10-16:
Here is an example of a file created from scratch
[image]
[test.FCStd.zip](https://github.com/realthunder/FreeCAD/files/5392212/test.FCStd.zip)

tiket18, 2020-10-16:
17:18:43  OpenGL version is: 4.6 (4.6 (Compatibility Profile) Mesa 20.2.0)
17:18:43  Run Gui init script
17:18:43  Init: Running FreeCADGuiInit.py start script...
17:18:43  Init:   Searching modules...
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/AddonManager... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Arch... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Draft... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Drawing... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Fem... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Idf(InitGui.py not found)... ignore
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Image... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Import... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Inspection... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Material... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Measure(InitGui.py not found)... ignore
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Mesh... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/MeshPart... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/OpenSCAD... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Part... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/PartDesign... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Path... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Points... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Raytracing... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/ReverseEngineering... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Robot... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Show(InitGui.py not found)... ignore
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Sketcher... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Spreadsheet... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Start... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Surface... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/TechDraw... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Test... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Tux... done
17:18:43  Init:      Initializing /usr/lib/freecad/Mod/Web... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/CADExchanger... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/dxf_library(InitGui.py not found)... ignore
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/Assembly4... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/FCGear(InitGui.py not found)... ignore
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/assembly2... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/Lithophane... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/CurvedShapes... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/frame(InitGui.py not found)... ignore
17:18:43  <Command> Command.cpp(1839): duplicate command Nurbs_TTcreateArcSketch
17:18:43  <Command> Command.cpp(1839): duplicate command Nurbs_TTcreateLabels
17:18:43  <Command> Command.cpp(1839): duplicate command Nurbs_TTupdateLabels
17:18:43  <Command> Command.cpp(1839): duplicate command Nurbs_weighteditor
17:18:43  <Command> Command.cpp(1839): duplicate command Nurbs_LoadSketch
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/nurbs... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/Curves... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/MeshRemodel... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/slic3r-tools... done
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/workfeature-macro(InitGui.py not found)... ignore
17:18:43  Init:      Initializing /home/memfis/.FreeCAD/Mod/A2plus... done
17:18:43  Init: Initializing freecad.asm3
17:18:43  QVTKWidgetPlugin instantiated

17:18:43  QVTKWidgetPlugin::name

17:18:43  1.312339 <asm3.main> init_gui.py(14): no solver backend found
17:18:43  Init: Initializing freecad.asm3... done
17:18:43  Init: Initializing freecad.fc_cadquery
17:18:43  Init: No init_gui module found in freecad.fc_cadquery, skipping
17:18:43  Init: Initializing freecad.frametools
17:18:43  Init: Initializing freecad.frametools... done
17:18:43  Init: Initializing freecad.gears
17:18:43  Init: Initializing freecad.gears... done
17:18:43  Init: Loading FreeCAD GUI
17:18:43  Init: Running FreeCADGuiInit.py start script... done
17:18:43  Init: Activating default workbench StartWorkbench
17:18:43  Loading GUI of Part module... done
17:18:43  Loading Sketcher module... done
17:18:43  Loading GUI of Sketcher module... done
17:18:43  Init: Showing main window
17:18:43  Main window restored
17:18:43  Show main window
17:18:43  Toolbars restored
17:18:43  Couldn't connect to spacenav daemon
17:18:43  Init: Entering event loop
17:18:43  Init: Processing command line files
17:18:43  Init:     Processing file: /home/memfis/трафареты2.FCStd
17:18:43  Cannot create object 'Sketch': (call stack is not deep enough)
17:18:43  Cannot create object 'Body': (call stack is not deep enough)
17:18:43  Cannot create object 'Pad': (call stack is not deep enough)
17:18:43  Cannot create object 'Sketch001': (call stack is not deep enough)
17:18:43  Cannot create object 'Pocket': (call stack is not deep enough)

tiket18, 2020-10-18:
11:27:49  OpenGL version is: 4.6 (4.6 (Compatibility Profile) Mesa 20.2.0)
11:27:49  Run Gui init script
11:27:49  Init: Running FreeCADGuiInit.py start script...
11:27:49  Init:   Searching modules...
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/AddonManager... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Arch... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Draft... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Drawing... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Fem... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Idf(InitGui.py not found)... ignore
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Image... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Import... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Inspection... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Material... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Measure(InitGui.py not found)... ignore
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Mesh... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/MeshPart... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/OpenSCAD... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Part... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/PartDesign... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Path... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Points... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Raytracing... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/ReverseEngineering... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Robot... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Show(InitGui.py not found)... ignore
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Sketcher... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Spreadsheet... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Start... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Surface... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/TechDraw... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Test... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Tux... done
11:27:49  Init:      Initializing /usr/lib/freecad/Mod/Web... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/CADExchanger... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/dxf_library(InitGui.py not found)... ignore
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/Assembly4... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/FCGear(InitGui.py not found)... ignore
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/assembly2... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/Lithophane... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/CurvedShapes... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/frame(InitGui.py not found)... ignore
11:27:49  <Command> Command.cpp(1839): duplicate command Nurbs_TTcreateArcSketch
11:27:49  <Command> Command.cpp(1839): duplicate command Nurbs_TTcreateLabels
11:27:49  <Command> Command.cpp(1839): duplicate command Nurbs_TTupdateLabels
11:27:49  <Command> Command.cpp(1839): duplicate command Nurbs_weighteditor
11:27:49  <Command> Command.cpp(1839): duplicate command Nurbs_LoadSketch
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/nurbs... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/Curves... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/MeshRemodel... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/slic3r-tools... done
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/workfeature-macro(InitGui.py not found)... ignore
11:27:49  Init:      Initializing /home/memfis/.FreeCAD/Mod/A2plus... done
11:27:49  Init: Initializing freecad.asm3
11:27:49  QVTKWidgetPlugin instantiated

11:27:49  QVTKWidgetPlugin::name

11:27:49  1.354572 <asm3.main> init_gui.py(14): no solver backend found
11:27:49  Init: Initializing freecad.asm3... done
11:27:49  Init: Initializing freecad.fc_cadquery
11:27:49  Init: No init_gui module found in freecad.fc_cadquery, skipping
11:27:49  Init: Initializing freecad.frametools
11:27:49  Init: Initializing freecad.frametools... done
11:27:49  Init: Initializing freecad.gears
11:27:49  Init: Initializing freecad.gears... done
11:27:49  Init: Loading FreeCAD GUI
11:27:49  Init: Running FreeCADGuiInit.py start script... done
11:27:49  Init: Activating default workbench StartWorkbench
11:27:49  Loading GUI of Part module... done
[... 91 lines cut ...]
Object: Body005
11:27:49  0.000594641 <App> Document.cpp(3113): _____________________________2#Link015.LinkedObject: Link not restored
Object: Body006
11:27:49  0.000755654 <App> Document.cpp(3113): _____________________________2#Link032.LinkedObject: Link not restored
Object: Fusion004
11:27:49  0.0008407 <App> Document.cpp(3113): _____________________________2#Link018.LinkedObject: Link not restored
Object: Body004
11:27:49  0.000868813 <App> Document.cpp(3113): _____________________________2#Link019.LinkedObject: Link not restored
Object: Body001
11:27:49  0.00089309 <App> Document.cpp(3113): _____________________________2#Link020.LinkedObject: Link not restored
Object: Body002
[... cut ...]
11:27:49  0.000988322 <App> Document.cpp(3113): _____________________________2#Link024.LinkedObject: Link not restored
Object: Body007
11:27:49  0.00101206 <App> Document.cpp(3113): _____________________________2#Link025.LinkedObject: Link not restored
Object: Body001
11:27:49  0.00103525 <App> Document.cpp(3113): _____________________________2#Link026.LinkedObject: Link not restored
Object: Body002
11:27:49  0.00105888 <App> Document.cpp(3113): _____________________________2#Link027.LinkedObject: Link not restored
Object: Body003

realthunder, 2020-10-18:
> 11:27:49 Cannot create object 'Sketch': (call stack is not deep enough)

This type of errors are the cause. But not sure why. Are you compiling on Linux? Can you run inside gdb? Before you open 
 the document. CTRL+C, and then type `catch throw`, then `continue`. Open the document. When the exception occur, it will break into the debugger. type `frame` and post the call stack here.

tiket18, 2020-10-18:
I don't have any experience with gdb - I installed it, ran it with gdb command, but I haven't understood yet how to run FC inside it.

But I've found out an interesting thing - the model file breaks only if you open the project file through the file manager (i.e. if you launch it with the command of -
freecad file.fcstd).
If you open the project file from the already running FC, through the Open file menu, everything works normally.

tiket18, 2020-10-21:
2020-10-21 build for today - the problem with opening a file remains. Run in command line = broken project (you need to open the file again through the Open menu).

OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25420 (Git)
Build type: Release
Branch: makepkg
Hash: dda4965b133f86ed98e03619a1891d58c8281fcb
Python version: 3.8.6
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.4.0
Locale: Russian/Russia (ru_RU)
freecad  лампа_настольная.FCStd
FreeCAD 0.19, Libs: 0.19R25420 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Loading Assembly4 WorkBench

(freecad:189785): Gtk-WARNING **: 12:34:36.794: Theme parsing error: gtk.css:4:108: Failed to import: Произошла ошибка открытия файла /home/m/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet.css: Нет такого файла или каталога

(freecad:189785): Gtk-WARNING **: 12:34:36.794: Theme parsing error: gtk.css:5:114: Failed to import: Произошла ошибка открытия файла /home/m/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet-tiled.css: Нет такого файла или каталога
<Command> Command.cpp(1839): duplicate command Std_TreePreSelection
<Command> Command.cpp(1839): duplicate command Std_TreeSelectAllInstances
1.25141 <asm3.main> init_gui.py(14): no solver backend found
connect failed: Нет такого файла или каталога
Cannot create object 'Sketch': (call stack is not deep enough)
Cannot create object 'Spreadsheet': (call stack is not deep enough)
Cannot create object 'Sketch065': (call stack is not deep enough)
Cannot create object 'Sketch066': (call stack is not deep enough)
Cannot create object 'Body013': (call stack is not deep enough)
Cannot create object 'Sketch067': (call stack is not deep enough)
Cannot create object 'Pad013': (call stack is not deep enough)
Cannot create object 'Sketch070': (call stack is not deep enough)
Cannot create object 'Sketch071': (call stack is not deep enough)
Cannot create object 'Pocket041': (call stack is not deep enough)
Cannot create object 'Sketch073': (call stack is not deep enough)
Cannot create object 'Pad015': (call stack is not deep enough)
Cannot create object 'Body015': (call stack is not deep enough)
Cannot create object 'Sketch072': (call stack is not deep enough)
Cannot create object 'Pad014': (call stack is not deep enough)
Cannot create object 'Body014': (call stack is not deep enough)
Cannot create object 'Part__Mirroring': (call stack is not deep enough)
Cannot create object 'Sketch064': (call stack is not deep enough)
Cannot create object 'Pad012': (call stack is not deep enough)
Cannot create object 'Body012': (call stack is not deep enough)
Cannot create object 'Sketch057': (call stack is not deep enough)
Cannot create object 'Sketch058': (call stack is not deep enough)
Cannot create object 'Sketch059': (call stack is not deep enough)
Cannot create object 'Sketch060': (call stack is not deep enough)
Cannot create object 'Sketch061': (call stack is not deep enough)
Cannot create object 'Sketch062': (call stack is not deep enough)
Cannot create object 'Body011': (call stack is not deep enough)
Cannot create object 'Sketch041': (call stack is not deep enough)
Cannot create object 'Sketch042': (call stack is not deep enough)
Cannot create object 'Sketch043': (call stack is not deep enough)
Cannot create object 'Pad009': (call stack is not deep enough)
Cannot create object 'Sketch044': (call stack is not deep enough)
Cannot create object 'Sketch046': (call stack is not deep enough)
Cannot create object 'Pocket040': (call stack is not deep enough)
[... 309 lines cut ...]
Object: Body
0.000212195 <App> Document.cpp(3113): _______________________________#Link003.LinkedObject: Link not restored
Object: Body009
0.000310712 <App> Document.cpp(3113): _______________________________#Link004.LinkedObject: Link not restored
Object: Body011
0.000394919 <App> Document.cpp(3113): _______________________________#Link005.LinkedObject: Link not restored
Object: Body012
0.000471344 <App> Document.cpp(3113): _______________________________#Link006.LinkedObject: Link not restored
Object: Part__Mirroring
0.000580003 <App> Document.cpp(3113): _______________________________#Link010.LinkedObject: Link not restored
Object: Body016
0.000669683 <App> Document.cpp(3113): _______________________________#Link017.LinkedObject: Link not restored
Object: Body007
0.000755356 <App> Document.cpp(3113): _______________________________#Link018.LinkedObject: Link not restored
Object: Body006
0.00125374 <App> Document.cpp(3113): _______________________________#Link019.LinkedObject: Link not restored
Object: Body024
0.00137234 <App> Document.cpp(3113): _______________________________#Link020.LinkedObject: Link not restored
Object: Body025
0.00171837 <App> Document.cpp(3113): _______________________________#Link021.LinkedObject: Link not restored
Object: Body011
0.00183156 <App> Document.cpp(3113): _______________________________#Link022.LinkedObject: Link not restored
Object: Body007
0.00192532 <App> Document.cpp(3113): _______________________________#Link023.LinkedObject: Link not restored
Object: Body006

tiket18, 2020-10-22:
The assembly from 22-10-2020 is better, but it still breaks -

OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25427 (Git)
Build type: Release
Branch: makepkg
Hash: d4401f23c304ec2de60e96f3fd77eccbe79b2139
Python version: 3.8.6
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.4.0
Locale: Russian/Russia (ru_RU)

<Command> Command.cpp(1839): duplicate command Std_TreePreSelection
<Command> Command.cpp(1839): duplicate command Std_TreeSelectAllInstances
1.283876 <asm3.main> init_gui.py(14): no solver backend found
connect failed: Нет такого файла или каталога
Cannot create object 'Body005': (call stack is not deep enough)
Cannot create object 'Sketch006': (call stack is not deep enough)
Cannot create object 'Pad005': (call stack is not deep enough)
Cannot create object 'Sketch007': (call stack is not deep enough)
Cannot create object 'Job001': (call stack is not deep enough)
Cannot create object 'Operations001': (call stack is not deep enough)
Cannot create object 'Clone005': (call stack is not deep enough)
Cannot create object 'T2__F3_175_15_z4005': (call stack is not deep enough)
Cannot create object 'Stock001': (call stack is not deep enough)
<class 'ValueError'>: call stack is not deep enough
<App> PropertyLinks.cpp(1033): Lost link to __________________3 Clone005 while loading, maybe an object was not loaded correctly
<App> PropertyLinks.cpp(1033): Lost link to __________________3 Sketch007 while loading, maybe an object was not loaded correctly
<class 'ValueError'>: call stack is not deep enough

realthunder, 2020-10-22:
This problem is hard to diagnose, as I cannot reproduce it on my computer. Might be some setting problem in your build environment?

tiket18, 2020-11-23:
Problem solved

OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25544 (Git)
Build type: Release
Branch: makepkg
Hash: c22fc9099d950f9ea8cf7b452656cb1ca4d68495
Python version: 3.8.6
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Russian/Russia (ru_RU)

closed 2020-11-23 by tiket18