realthunder/FreeCAD #54: default units of measurement and display of names after renaming

tiket18, 2020-10-20:
С некоторых пор внезапно стало требоваться и в таблице, и в задании ограничений через формулы явно указывать единицы измерения. Соответственно возникла проблема - если ранее я где-нибудь сделал ограничение типа B1+20 то при первом же изменении в эскизе все ломается - он начинает требовать, чтобы я прошел по всем таким местам и явно задал, что 20 это в мм. Было бы правильно, чтобы он сам это понимал везде и всегда (как было раньше) - ведь мы в настройках указываем значения мм/кг и пр.

Побочная проблема -  с отображением. Очень хорошо, что мы имеем возможность переименовать эскизы.  Но этим создаем себе проблему в будущем - когда решатель решает на что-нибудь пожаловаться, он пишет жалобу не на новое имя эскиза, а на исходное, вида SketcherNN и в дереве проекта это SketcherNN не найти

Since some time, it has suddenly become necessary to explicitly specify units of measurement in the table and in the assignment of restrictions through formulas. Accordingly, there was a problem - if earlier I somewhere made a restriction of type B1+20 then at the first change in the sketch all breaks - he begins to demand that I went through all such places and clearly set that 20 is in mm. It would be right for him to understand it everywhere and always (as it used to be), because we specify the values of mm/kg in the settings, etc.

A side problem is with the display. It's very good that we can rename the thumbnails.  But this creates a problem in the future - when the solver decides to complain about something, he writes a complaint not on the new sketch name, but on the original, the kind of SketcherNN and in the project tree it is SketcherNNN not find

tiket18, 2020-10-20:
It only shows up in the status bar but I (personally) turn it off on narrow screens and there is very little room.

tiket18, 2020-10-20:
[image]
Here is an example. It clearly indicates an error in the size setting in the sketch 73. What exactly is the constraint? - does not indicate. Here, the tree shows absolutely all specified limitations - all dimensions are specified correctly in them.

tiket18, 2020-10-20:
Yay, I have caught this moment. The problem is checked both in the current build of this branch of appimage, and in the one I have built locally. The problem is not reproduced in the main branch. It looks like it - here I see a supposedly broken sketch in the project tree (due to complaints of the solver not specified units of measurement), open this sketch and check it - all the constraints that are set with the value specified correctly. But suddenly I see - there are restrictions without knowledge, but for some reason they have the "calculated by function" icon on the right - in this case it is for some reason a horizontal restriction. I can delete it, set a new one = the program (what is locally assembled, what is the apimage) immediately drops. The main branch silently closes the sketch and sees no more errors in it.
[image]
<Sketch> SketchObject.cpp(6906): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g296;SKT
<Sketch> SketchObject.cpp(6906): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g297;SKT
<Sketch> SketchObject.cpp(6906): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g302;SKT
<Sketch> SketchObject.cpp(6906): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g288;SKT
<Sketch> SketchObject.cpp(6906): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g293;SKT
<Sketch> SketchObject.cpp(6906): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g321;SKT
<Sketch> SketchObject.cpp(6957): External geometry _______________________________#Sketch067.e355 missing reference: ���c<Sketch> SketchObject.cpp(6957): External geometry _______________________________#Sketch067.e356 missing reference: �����<Sketch> SketchObject.cpp(6957): External geometry _______________________________#Sketch067.e357 missing reference: $J��<Sketch> SketchObject.cpp(6957): External geometry _______________________________#Sketch067.e358 missing reference: <Sketch> SketchObject.cpp(6957): External geometry _______________________________#Sketch067.e359 missing reference: � �q<VProgram received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d6a0) [0x7fc5b70266a0]
#1  0x7fc56c46653f in Sketcher::SketchObject::rebuildVertexIndex() from /usr/lib/freecad/lib/Sketcher.so+0x14f
#2  0x7fc56c469441 in Sketcher::SketchObject::rebuildExternalGeometry(bool) from /usr/lib/freecad/lib/Sketcher.so+0xb31
#3  0x7fc56c47b303 in Sketcher::SketchObject::execute() from /usr/lib/freecad/lib/Sketcher.so+0x63
#4  0x7fc5b8f4453a in App::DocumentObject::recompute() from /usr/lib/freecad/lib/libFreeCADApp.so+0x8a
#5  0x7fc5b1876f6a in Part::Feature::recompute() from /usr/lib/freecad/lib/Part.so+0xa
#6  0x7fc5b8ecee29 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x219
#7  0x7fc5b8ee6567 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad/lib/libFreeCADApp.so+0x997
#8  0x7fc5b8f9d09f in App::DocumentPy::recompute(_object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x2df
#9  0x7fc5b8f9d297 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x57
#10  /usr/lib/libpython3.8.so.1.0(PyCFunction_Call+0x19a) [0x7fc5b87928fa]
#11  /usr/lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x464) [0x7fc5b87854d4]
#12  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x50e8) [0x7fc5b8780da8]
#13  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x304) [0x7fc5b877a9a4]
#14  /usr/lib/libpython3.8.so.1.0(PyEval_EvalCode+0x23) [0x7fc5b882af03]
#15  /usr/lib/libpython3.8.so.1.0(+0x1d8868) [0x7fc5b8836868]
#16  /usr/lib/libpython3.8.so.1.0(+0x1d2a43) [0x7fc5b8830a43]
#17  /usr/lib/libpython3.8.so.1.0(PyRun_StringFlags+0x7d) [0x7fc5b882d8ad]
#18  0x7fc5b8b3fa0c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x6c
#19  0x7fc5b987c716 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x96
#20  0x7fc5b987c896 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib/freecad/lib/libFreeCADGui.so+0xe6
#21  0x7fc5b987c9c5 in Gui::Command::updateActive() from /usr/lib/freecad/lib/libFreeCADGui.so+0x45
#22  0x7fc56c76188d in SketcherGui::ViewProviderSketch::unsetEdit(int) from /usr/lib/freecad/lib/SketcherGui.so+0x71d
#23  0x7fc5b9c0a7a8 in Gui::ViewProvider::finishEditing() from /usr/lib/freecad/lib/libFreeCADGui.so+0x38
#24  0x7fc5b97e6e3d in Gui::Document::_resetEdit() from /usr/lib/freecad/lib/libFreeCADGui.so+0x9d
#25  0x7fc5b978ef1a in Gui::Application::setEditDocument(Gui::Document*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x3a
#26  0x7fc5b97b99ff in Gui::Application::sResetEdit(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x2f
#27  /usr/lib/libpython3.8.so.1.0(PyCFunction_Call+0x19a) [0x7fc5b87928fa]
#28  /usr/lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x464) [0x7fc5b87854d4]
#29  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x50e8) [0x7fc5b8780da8]
#30  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x304) [0x7fc5b877a9a4]
[... cut ...]
#64  0x7fc5b76b5941 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x61
#65  0x7fc5b765b65c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12c
#66  0x7fc5b7663af4 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x94
#67  0x7fc5b979a9cc in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x181c
#68  freecad(+0x45e0) [0x563c607c55e0]
#69  /usr/lib/libc.so.6(__libc_start_main+0xf2) [0x7fc5b7011152]
#70  freecad(+0x48ee) [0x563c607c58ee]

realthunder, 2020-10-21:
Can you please post a file showing the problem? Things like `1 + 2mm` is considered illegal because of mixed units. If it is `1 + 2`, then it will be default to `mm` if bound to a quantity property.

Regarding the object name inside error message, you can easily find the object by pressing `CTRL + F` in tree view, and type the name you see in the message. The reason to not show label (the user assigned name) is because of the message may show object hierarchy. So a user changeable label containing arbitrary character will be confusing. Note that you can directly type the object hierarchy in the search box to find the child item.

tiket18, 2020-10-21:
файл выслал Вам на электронную почту - гитхаб не пускает по размеру.
file was sent to you by e-mail - githab does not allow the size.

tiket18, 2020-10-21:
It wasn't me who made it up about mixed indication of units. The problem arose when in the table I began not only to set the numbers, and in the neighboring cells to write formulas like B11 +2 * B12 +2 and in such cases in +2 was clearly required to specify mm. I think this league should be corrected.

tiket18, 2020-10-22:
The assembly from 22-10-2020, is better, opening and closing such a sketch solves the problem. Thank you very much for the progress, for your work.

closed 2020-10-22 by tiket18