realthunder/FreeCAD #55: The program often drops when trying to change some restriction, to save the change.

tiket18, 2020-10-22:
This is not a new problem, it has been around for quite a while. I put up with it (just saving often). Looks like this - I enter the sketch which the solver means as "there are errors". I see, I remove the restriction = may fall. I run it again, open it, delete it = may not fall. The difficulty of catching is that I rarely launch the program from the console (where after a crash I can see the error messages). Here is an example - sketch067 rules, I add a horizontal restriction for the upper horizontal line (file is the same as the one I sent by email) = write "sketch is fully defined", press Ctrl+s (save)  and (or) 
[image]
 exit the sketch = fall.

<Command> Command.cpp(1839): duplicate command Std_TreePreSelection
<Command> Command.cpp(1839): duplicate command Std_TreeSelectAllInstances
1.3231 <asm3.main> init_gui.py(14): no solver backend found
connect failed: Нет такого файла или каталога
<Sketch> SketchObject.cpp(6905): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g296;SKT
<Sketch> SketchObject.cpp(6905): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g297;SKT
<Sketch> SketchObject.cpp(6905): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g302;SKT
<Sketch> SketchObject.cpp(6905): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g288;SKT
<Sketch> SketchObject.cpp(6905): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g293;SKT
<Sketch> SketchObject.cpp(6905): Duplicated external reference in _______________________________#Sketch067: Sketch066.;g321;SKT
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d6a0) [0x7f9e43c586a0]
#1  0x7f9df90a1658 in Sketcher::SketchObject::rebuildExternalGeometry(bool) from /usr/lib/freecad/lib/Sketcher.so+0x1db8
#2  0x7f9df90b2003 in Sketcher::SketchObject::execute() from /usr/lib/freecad/lib/Sketcher.so+0x63
#3  0x7f9e45b764ea in App::DocumentObject::recompute() from /usr/lib/freecad/lib/libFreeCADApp.so+0x8a
#4  0x7f9e3e4a8f6a in Part::Feature::recompute() from /usr/lib/freecad/lib/Part.so+0xa
#5  0x7f9e45b00dd9 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x219
#6  0x7f9e45b18517 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad/lib/libFreeCADApp.so+0x997
#7  0x7f9e45bcf04f in App::DocumentPy::recompute(_object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x2df
#8  0x7f9e45bcf247 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x57
#9  /usr/lib/libpython3.8.so.1.0(PyCFunction_Call+0x19a) [0x7f9e453c48fa]
#10  /usr/lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x464) [0x7f9e453b74d4]
#11  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x50e8) [0x7f9e453b2da8]
#12  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x304) [0x7f9e453ac9a4]
#13  /usr/lib/libpython3.8.so.1.0(PyEval_EvalCode+0x23) [0x7f9e4545cf03]
#14  /usr/lib/libpython3.8.so.1.0(+0x1d8868) [0x7f9e45468868]
#15  /usr/lib/libpython3.8.so.1.0(+0x1d2a43) [0x7f9e45462a43]
#16  /usr/lib/libpython3.8.so.1.0(PyRun_StringFlags+0x7d) [0x7f9e4545f8ad]
#17  0x7f9e45771a0c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x6c
#18  0x7f9e464ae876 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x96
#19  0x7f9e464ae9f6 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib/freecad/lib/libFreeCADGui.so+0xe6
#20  0x7f9e464aeb25 in Gui::Command::updateActive() from /usr/lib/freecad/lib/libFreeCADGui.so+0x45
#21  0x7f9df93998ad in SketcherGui::ViewProviderSketch::unsetEdit(int) from /usr/lib/freecad/lib/SketcherGui.so+0x71d
#22  0x7f9e4683cf68 in Gui::ViewProvider::finishEditing() from /usr/lib/freecad/lib/libFreeCADGui.so+0x38
#23  0x7f9e46418f1d in Gui::Document::_resetEdit() from /usr/lib/freecad/lib/libFreeCADGui.so+0x9d
#24  0x7f9e463c101a in Gui::Application::setEditDocument(Gui::Document*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x3a
#25  0x7f9e463ebaff in Gui::Application::sResetEdit(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x2f
#26  /usr/lib/libpython3.8.so.1.0(PyCFunction_Call+0x19a) [0x7f9e453c48fa]
#27  /usr/lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x464) [0x7f9e453b74d4]
#28  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x50e8) [0x7f9e453b2da8]
#29  /usr/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x304) [0x7f9e453ac9a4]
#30  /usr/lib/libpython3.8.so.1.0(PyEval_EvalCode+0x23) [0x7f9e4545cf03]
#31  /usr/lib/libpython3.8.so.1.0(+0x1d8868) [0x7f9e45468868]
#32  /usr/lib/libpython3.8.so.1.0(+0x1d2a43) [0x7f9e45462a43]
#33  /usr/lib/libpython3.8.so.1.0(PyRun_StringFlags+0x7d) [0x7f9e4545f8ad]
#34  0x7f9e45771a0c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x6c
#35  0x7f9e464ae876 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x96
#36  0x7f9e464ae9f6 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib/freecad/lib/libFreeCADGui.so+0xe6
#37  0x7f9df93c813b in SketcherGui::TaskDlgEditSketch::reject() from /usr/lib/freecad/lib/SketcherGui.so+0xfb
#38  0x7f9e4679ff6c in Gui::TaskView::TaskView::reject() from /usr/lib/freecad/lib/libFreeCADGui.so+0x5c
#39  /usr/lib/libQt5Core.so.5(+0x2eb070) [0x7f9e442c6070]
#40  /usr/lib/libQt5Widgets.so.5(+0x2f2291) [0x7f9e44eee291]
#41  /usr/lib/libQt5Core.so.5(+0x2eb070) [0x7f9e442c6070]
#42  0x7f9e44e47b23 in QAbstractButton::clicked(bool) from /usr/lib/libQt5Widgets.so.5+0x43
#43  /usr/lib/libQt5Widgets.so.5(+0x24c4ac) [0x7f9e44e484ac]
#44  /usr/lib/libQt5Widgets.so.5(+0x24de53) [0x7f9e44e49e53]
#45  0x7f9e44e4a033 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt5Widgets.so.5+0xf3
#46  0x7f9e44d97b0e in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x1ce
#47  0x7f9e44d56752 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x82
#48  0x7f9e44d5d87b in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x4ab
#49  0x7f9e464785f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x89
#50  0x7f9e4428ecda in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x12a
#51  0x7f9e44d5c87e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt5Widgets.so.5+0x1de
#52  /usr/lib/libQt5Widgets.so.5(+0x1b4249) [0x7f9e44db0249]
#53  /usr/lib/libQt5Widgets.so.5(+0x1b763f) [0x7f9e44db363f]
#54  0x7f9e44d56752 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x82
#55  0x7f9e464785f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x89
#56  0x7f9e4428ecda in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x12a
#57  0x7f9e446694ac in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt5Gui.so.5+0x6ac
#58  0x7f9e4463ebac in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Gui.so.5+0xbc
#59  /usr/lib/libQt5XcbQpa.so.5(+0x5ef9c) [0x7f9e3a97cf9c]
#60  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x184) [0x7f9e41fbb924]
#61  /usr/lib/libglib-2.0.so.0(+0xa6621) [0x7f9e4200f621]
#62  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f9e41fba131]
#63  0x7f9e442e7941 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x61
#64  0x7f9e4428d65c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12c
#65  0x7f9e44295af4 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x94
#66  0x7f9e463ccacc in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x181c
#67  freecad(+0x45e0) [0x55ec21b3d5e0]
#68  /usr/lib/libc.so.6(__libc_start_main+0xf2) [0x7f9e43c43152]
#69  freecad(+0x48ee) [0x55ec21b3d8ee]

realthunder, 2020-10-22:
Do you still have this problem in the latest build? I did fix one bug causing potential crash in `rebuildExternalGeometry()`.

tiket18, 2020-10-22:
Yes, in the apimage such a problem on this particular file does not work. Thank you.  We will see.

luzpaz, 2022-01-19:
@tiket18 still reproducible on latest LinkDaily?

closed 2022-01-19 by tiket18