realthunder/FreeCAD #56: Does not want to move objects to groups

tiket18, 2020-10-22:
[image]
OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.10.22.22739 +2688 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: d4401f23c304ec2de60e96f3fd77eccbe79b2139
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Russian/Russia (ru_RU)

00:36:02  Loading PathDressup... done
00:36:02  Loading PathDrillingGui... done
00:36:02  Loading PathEngraveGui... done
00:36:02  Loading PathFixture... done
00:36:02  Loading PathHelixGui... done
00:36:02  Loading PathHop... done
00:36:02  Loading PathMillFaceGui... done
00:36:02  Loading PathPocketGui... done
00:36:02  Loading PathPocketShapeGui... done
00:36:02  Loading PathPost... done
00:36:02  Loading PathProbeGui... done
00:36:02  Loading PathProfileFacesGui... done
00:36:02  Loading Mesh module... done
00:36:02  Loading PathSimulator module.... done
00:36:02  Loading PathSimulator Gui... done
00:36:02  Loading PathSlotGui... done
00:36:02  Loading PathStop... done
00:36:02  Loading PathVcarveGui... done
00:36:03  4.61e-07 <App> Document.cpp(3439): The graph must be a DAG.
00:36:03  0.0155358 <App> Document.cpp(3439): The graph must be a DAG.
00:36:13  10.3524 <App> Document.cpp(3404): Dependency cycles: 
00:36:13  10.3524 <App> Document.cpp(3434): 

_________________________________#Sketch, _________________________________#Spreadsheet, _________________________________#Sketch005, 

00:36:13  <Exception> Document.cpp(3435): Cyclice dependency detected.
Please check Report View for more details.

[GCC 7.5.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',0)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> Gui.runCommand('Std_ViewStatusBar',0)
>>> ### Begin command Std_Open
>>> FreeCAD.openDocument('/home/memfis/H/A_w/\u0422\u0435\u043c\u044b/2017/rozetka/\u0441\u0432\u0435\u0442\u0438\u043b\u044c\u043d\u0438\u043a-\u0432\u0438\u0442\u0440\u0430\u0436.FCStd')
>>> # App.setActiveDocument("_________________________________")
>>> # App.ActiveDocument=App.getDocument("_________________________________")
>>> # Gui.ActiveDocument=Gui.getDocument("_________________________________")
>>> ### End command Std_Open
>>> # Gui.Selection.addSelection('_________________________________','Job')
>>> # Gui.Selection.addSelection('_________________________________','Job001')
>>> # Gui.Selection.addSelection('_________________________________','Job002')
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('_________________________________','Job002')
>>> # Gui.Selection.addSelection('_________________________________','Job')
>>> # Gui.Selection.addSelection('_________________________________','Job001')
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('_________________________________','____________001')
>>> App.getDocument('_________________________________').getObject('Job').adjustRelativeLinks(App.getDocument('_________________________________').getObject('____________001'))
>>> App.getDocument("_________________________________").getObject("____________001").addObject(App.getDocument("_________________________________").getObject("Job"))
>>> Gui.runCommand('Std_WhatsThis',0)
>>> Gui.runCommand('Std_About',0)
>>> 
[image]
[ewvfew.FCStd.zip](https://github.com/realthunder/FreeCAD/files/5426009/ewvfew.FCStd.zip)

realthunder, 2020-10-26:
The error is cause by dependency loop, as shown in the error message. The loop is formed by three object, 

`Sketch, Spreadsheet, Sketch005`

`Sketch` reference `Spreadsheet` in its constraint
`Spreadsheet` reference `Sketch005.Shape.Edge2.Length` in `B17`
`Sketch005` reference `Sketch` in its external (i.e. imported) geometry.

There is a way to work around the cyclic dependency reference. Use the `href()` function, `href` stands for `hidden reference`. Be warned though, this is an advanced feature. The dependency loop will cause unexpected recomputation problems, because of unstable recomputation order.

`href(Sketch.Shape.Edge2.Length)`

Also beware that direct referencing to Sub-element of a `Shape` will get you in trouble with topological renaming when the shape is modified. Use the following syntax for more robust referencing,

`href(Sketch.<<Edge2>>._shape.Length)`

realthunder, 2020-10-26:
Forgot to mention. The recommended way of solving dependency loop here is to split the Spreadsheet into two if possible.

luzpaz, 2022-01-19:
@realthunder anything left to do here?

closed 2022-01-19 by tiket18