realthunder/FreeCAD #59: Bug: Multi-Body Pad missing elements

obelisk79, 2020-11-02:
I have created a multi-element sketch to recreate a 'polar-array' for ultimate use as part of a cycloidal drive speed reducer. Here is my sketch and resulting solid after applying a pad.
[image]
[image]

I've also attached the FCStd file so you can check and verify it isn't just an error on my part.
[cycloidal_drive_V1.zip](https://github.com/realthunder/FreeCAD/files/5477394/cycloidal_drive_V1.zip)

obelisk79, 2020-11-02:
Ok, I erased and recreated the sketch with slightly different constraints/construction lines and got proper geometry so not sure if it is an actual bug or just poor sketch on my part. The behavior seems odd. I've attached the updated file with working sketch/pad for your review.

[cycloidal_drive_V1.1.FCStd.zip](https://github.com/realthunder/FreeCAD/files/5477512/cycloidal_drive_V1.1.FCStd.zip)

realthunder, 2020-11-03:
There is probably something wrong with sketcher. Looking close at your original sketcher (when not in edit), you'll find two extra vertexes.

[image]

These turn out to be two circles with zero radius. So the bug in sketcher is that it allows you to create such circle. Maybe try to remember how did you get these circles? You can use the sketch task panel to select and delete those degenerate edges.

[image]

obelisk79, 2020-11-03:
Thanks for the response, that makes perfect sense. I may have accidentally double-clicked when attempting to create all of the circles which created the extra geometry without realizing what I had done.

closed 2020-11-03 by obelisk79