realthunder/FreeCAD #63: duplicate command

dulouie, 2020-11-06:
I compile my linkstage3 branch myself from the sources and on every start the following messages appear:
```
Tried to run Gui::Application::initApplication() twice!
<Command> Command.cpp(1839): duplicate command Std_TreePreSelection
<Command> Command.cpp(1839): duplicate command Std_TreeSelectAllInstances
```

snelweg, 2020-12-05:
It's still here. I start the appimage in a terminal window and see that message always. 
`/tmp/.mount_FreeCA6IomlE
FreeCAD 0.19, Libs: 2020.12.03R23076 +2885 (Git)
Sheet Metal workbench loaded
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
Traceback (most recent call last):
  File "<string>", line 239, in onStart
  File "<string>", line 228, in onWorkbenchActivated
AssertionError: No active workbench
`

ceremcem, 2020-12-05:
I can confirm that the messages exist within the current latest commit: 

```
FreeCAD 0.19, Libs: 0.19R25975 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
connect failed: No such file or directory
```

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25975 (Git)
Build type: Release
Branch: LinkStage3
Hash: 59bd7ff5a1c124eb1ab46d7b667a1ea3c5cca3ea
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

Built with[¹](https://github.com/ceremcem/build-freecad-asm3/tree/c37fa68a53ef5f012aa35e389737b834c95d1eb0): 

```
(cd $build_dir && cmake ../../FreeCAD \
	-DFREECAD_USE_OCC_VARIANT="Official Version" \
	-DCMAKE_BUILD_TYPE=Release \
	-DBUILD_QT5=ON \
	-DPYTHON_EXECUTABLE=/usr/bin/python3 \
)
```

commit 06c16d8e (2020-12-06): Gui: fix duplicate command

Fixes #63
closed 2020-12-06 by realthunder