realthunder/FreeCAD #64: boolean spli

dulouie, 2020-11-06:


dulouie, 2020-11-06:
And increasing the Extrusion plane length cause crash.

realthunder, 2020-11-08:
The split failure is most likely an OCC problem. You can see that the split is not correct near the seam edge of the cylinder. You can work around this by split at a different angle, avoiding the seam edge.

[image]

I can't reproduce the crash reliably. I did fix a crash problem related to rendering. Not sure if this is the crash you experienced. Did you have any stack trace when it crashed?

dulouie, 2020-11-08:
At 153b298a1cb52f5040fdc0ac371173586d2a2686 increasing and decreasing the length of the extrusion multiple times i get this crash log.

> Tried to run Gui::Application::initApplication() twice!
<Command> Command.cpp(1839): duplicate command Std_TreePreSelection
<Command> Command.cpp(1839): duplicate command Std_TreeSelectAllInstances
connect failed: Datei oder Verzeichnis nicht gefunden
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f0b45cb4210]
#1  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x2c6f35) [0x7f0b32c48f35]
#2  0x7f0a9a534f52 in PartGui::SoBrepFaceSet::renderFaces(SoCoordinateElement const*, int const*, int, int const*, int, int, SbVec3f const*, int const*, SoMaterialBundle*, int const*, SoTextureCoordinateBundle&, int const*, int, int, int) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0x1302
#3  0x7f0a9a539df3 in PartGui::SoBrepFaceSet::renderShape(SoGLRenderAction*) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0x363
#4  0x7f0a9a53bcd8 in PartGui::SoBrepFaceSet::renderShape(SoGLRenderAction*, std::shared_ptr<Gui::SoFCSelectionContextEx>, std::shared_ptr<Gui::SoFCSelectionContextEx>, bool, bool) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0x3f8
#5  0x7f0a9a53dace in PartGui::SoBrepFaceSet::glRender(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0x17ee
#6  0x7f0b458aad02 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x3d2
#7  0x7f0b458aad02 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x3d2
#8  0x7f0b456af277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#9  0x7f0b458464d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#10  0x7f0b4890ae1f in Gui::SoFCSwitch::doAction(SoAction*) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x75f
#11  0x7f0b458aa1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#12  0x7f0b48909b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#13  0x7f0b4890a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#14  0x7f0b4589144a in SoGroup::GLRender(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xca
#15  0x7f0b456af277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#16  0x7f0b458463f7 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x87
#17  0x7f0b4890abd5 in Gui::SoFCSwitch::doAction(SoAction*) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x515
#18  0x7f0b458aa1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#19  0x7f0b48909b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#20  0x7f0b4890a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#21  0x7f0b458aa1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#22  0x7f0b48909b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#23  0x7f0b4890a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#24  0x7f0b456af277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#25  0x7f0b458463f7 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x87
#26  0x7f0b4890abd5 in Gui::SoFCSwitch::doAction(SoAction*) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x515
#27  0x7f0b458aa1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#28  0x7f0b48909b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#29  0x7f0b4890a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#30  0x7f0b458aa1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#31  0x7f0b48906429 in Gui::SoFCPathAnnotation::GLRenderBelowPath(SoGLRenderAction*) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x489
#32  0x7f0b458aa1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
[... cut ...]
#83  0x7f0b462e14db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#84  0x7f0b462e9246 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#85  0x7f0b485c0b0c in Gui::Application::runApplication() from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x166c
#86  ./bin/FreeCAD(+0x4c8e) [0x5605fe84ac8e]
#87  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f0b45c950b3]
#88  ./bin/FreeCAD(+0x501e) [0x5605fe84b01e]

dulouie, 2020-11-08:
With up-to-date branch at a386a1256ec3aa898f08a3bb0ca57e9dd8a62a9c i get also a crash after some time of fiddling with the extrusion length.

> Tried to run Gui::Application::initApplication() twice!
<Command> Command.cpp(1839): duplicate command Std_TreePreSelection
<Command> Command.cpp(1839): duplicate command Std_TreeSelectAllInstances
connect failed: Datei oder Verzeichnis nicht gefunden
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
<Exception> ShapeBinder.cpp(489): Missing support object
Recompute failed! Please check report view.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f895b514210]
#1  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x2c6f35) [0x7f894d4e2f35]
#2  0x7f88aff63b12 in PartGui::SoBrepFaceSet::renderFaces(SoCoordinateElement const*, int const*, int, int const*, int, int, SbVec3f const*, int const*, SoMaterialBundle*, int const*, SoTextureCoordinateBundle&, int const*, int, int, int) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0xec2
#3  0x7f88aff68e03 in PartGui::SoBrepFaceSet::renderShape(SoGLRenderAction*) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0x363
#4  0x7f88aff6ace8 in PartGui::SoBrepFaceSet::renderShape(SoGLRenderAction*, std::shared_ptr<Gui::SoFCSelectionContextEx>, std::shared_ptr<Gui::SoFCSelectionContextEx>, bool, bool) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0x3f8
#5  0x7f88aff6cade in PartGui::SoBrepFaceSet::glRender(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/Mod/Part/PartGui.so+0x17ee
#6  0x7f895b10ad02 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x3d2
#7  0x7f895b10ad02 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x3d2
#8  0x7f895af0f277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#9  0x7f895b0a64d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#10  0x7f895e16ae1f in Gui::SoFCSwitch::doAction(SoAction*) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x75f
#11  0x7f895b10a1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#12  0x7f895e169b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#13  0x7f895e16a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#14  0x7f895b0f144a in SoGroup::GLRender(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xca
#15  0x7f895af0f277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#16  0x7f895b0a63f7 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x87
#17  0x7f895e16abd5 in Gui::SoFCSwitch::doAction(SoAction*) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x515
#18  0x7f895b10a1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#19  0x7f895e169b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#20  0x7f895e16a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#21  0x7f895b10a1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#22  0x7f895e169b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#23  0x7f895e16a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#24  0x7f895af0f277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#25  0x7f895b0a63f7 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x87
#26  0x7f895e16abd5 in Gui::SoFCSwitch::doAction(SoAction*) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x515
#27  0x7f895b10a1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
#28  0x7f895e169b6e in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x8ae
#29  0x7f895e16a0b4 in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0xd4
#30  0x7f895b10a1ce in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x4be
[... cut ...]
#83  0x7f895bb414db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#84  0x7f895bb49246 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#85  0x7f895de20b0c in Gui::Application::runApplication() from /home/arne/src/LinkStage3/build/lib/libFreeCADGui.so+0x166c
#86  ./bin/FreeCAD(+0x4c8e) [0x55d6d08d5c8e]
#87  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f895b4f50b3]
#88  ./bin/FreeCAD(+0x501e) [0x55d6d08d601e]

dulouie, 2020-11-08:
But it's strange sometimes it happens and sometime not :D

luzpaz, 2022-01-19:
@dulouie any updates on this?