realthunder/FreeCAD #67: FEM not working

tiket18, 2020-11-20:
OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.11.16.22739 +2766 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 6730ad3b3bd841c2745ec98bf953c523dea86913
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Russian/Russia (ru_RU)

Refuses to create a grid. Judging by the report and processor load - creates. But it does not show, does not show.

davidosterberg, 2020-12-12:
I have the same issue.

With LinkStage3 (2020-12-08 AppImage)
[image]
With standard FreeCAD 0.19. it works
[image]

**LinkStage about info**
OS: Ubuntu 20.10 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.08.23076 +2912 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 1eba27283118bae43f5264d38caf7451b848f49a
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

**

Vanilla FreeCAD about info**
OS: Ubuntu 20.10 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23058 (Git) AppImage
Build type: Release
Branch: master
Hash: 6e60870ebcb2df018adb7a1281c00dde87750f11
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Example file: (Remove zip filename prefix)
[link_fem_bug.FCStd.zip](https://github.com/realthunder/FreeCAD/files/5683848/link_fem_bug.FCStd.zip)

davidosterberg, 2020-12-12:
@berndhahnebach do you know what might be causing this?

realthunder, 2020-12-13:
This is caused by missing installation of gmsh command line in my release. It's been fixed now. Will make a new release soon.

closed 2020-12-13 by realthunder