realthunder/FreeCAD #72: Draft WB: "UI file does not exist"

ceremcem, 2020-11-29:
I compiled the LinkStage3 branch with the following `cmake` arguments: 

```
        cmake ../../FreeCAD \                                                                                                                                          
                -DCMAKE_INSTALL_PREFIX:PATH=$FREECAD \                                                                                                                 
                -DOCC_INCLUDE_DIR=$FREECAD/include/opencascade \                                                                                                       
                -DFREECAD_USE_OCC_VARIANT="Official Version" \                                                                                                         
                -DOpenCASCADE_DIR=$FREECAD/lib/cmake/opencascade \                                                                                                     
                -DCMAKE_BUILD_TYPE=$build_type \                                                                                                                       
                -DBUILD_QT5=ON \                                                                                                                                       
                -DPYTHON_EXECUTABLE=/usr/bin/python3 
```

Version information of the result executable is as follows: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25935 (Git)
Build type: Release
Branch: LinkStage3
Hash: a491be7f04516e65647d77ceb2547fbd0adac1df
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

However, there are a few abnormal things here. First thing is that the following error is shown while FreeCAD is invoked by the command line: 
```
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
During initialization the error "UI file does not exist" occurred in /opt/FreeCAD/Mod/Draft/InitGui.py
Please look into the log file for further information
During initialization the error "UI file does not exist" occurred in /opt/FreeCAD/Mod/Arch/InitGui.py
Please look into the log file for further information
During initialization the error "'PySide2.QtCore.qRegisterResourceData' called with wrong argument types:
  PySide2.QtCore.qRegisterResourceData(int, str, str, str)
Supported signatures:
  PySide2.QtCore.qRegisterResourceData(int, bytes, bytes, bytes)" occurred in /opt/FreeCAD/Mod/Ship/InitGui.py
Please look into the log file for further information
During initialization the error "'PySide2.QtCore.qRegisterResourceData' called with wrong argument types:
  PySide2.QtCore.qRegisterResourceData(int, str, str, str)
Supported signatures:
  PySide2.QtCore.qRegisterResourceData(int, bytes, bytes, bytes)" occurred in /opt/FreeCAD/Mod/Plot/InitGui.py
Please look into the log file for further information
5.033291 <asm3.main> init_gui.py(14): no solver backend found
Cannot find icon: Draft_Dimension
connect failed: No such file or directory
Traceback (most recent call last):
  File "<string>", line 118, in Initialize

UI file does not exist
```

A similar set of messages printed when I attempt to switch to Draft WB: 

```
Cannot find icon: Draft_Line
Cannot find icon: Draft_Wire
Cannot find icon: Draft_Fillet
Cannot find icon: Draft_Arc
Cannot find icon: Draft_Arc_3Points
Cannot find icon: Draft_Circle
Cannot find icon: Draft_Ellipse
Cannot find icon: Draft_Rectangle
Cannot find icon: Draft_Polygon
Cannot find icon: Draft_BSpline
Cannot find icon: Draft_CubicBezCurve
Cannot find icon: Draft_BezCurve
Cannot find icon: Draft_Point
Cannot find icon: Draft_Facebinder
Cannot find icon: Draft_ShapeString
Cannot find icon: Draft_Text
Cannot find icon: Draft_Dimension
Cannot find icon: Draft_Label
Cannot find icon: :icons/Draft_Annotation_Style.svg
Cannot find icon: Draft_Move
Cannot find icon: Draft_Rotate
Cannot find icon: Draft_Scale
Cannot find icon: Draft_Mirror
Cannot find icon: Draft_Offset
Cannot find icon: Draft_Trimex
Cannot find icon: Draft_Stretch
Cannot find icon: Draft_Clone
Cannot find icon: Draft_Array
Cannot find icon: Draft_PolarArray
Cannot find icon: Draft_CircularArray
Cannot find icon: Draft_PathArray
Cannot find icon: Draft_PathLinkArray
Cannot find icon: Draft_PointArray
Cannot find icon: Draft_PointLinkArray
Cannot find icon: Draft_PathTwistedArray
Cannot find icon: Draft_PathTwistedLinkArray
Cannot find icon: Draft_Array
Cannot find icon: Draft_Edit
Cannot find icon: Draft_SubelementHighlight
Cannot find icon: Draft_Join
Cannot find icon: Draft_Split
Cannot find icon: Draft_Upgrade
Cannot find icon: Draft_Downgrade
Cannot find icon: Draft_WireToBSpline
Cannot find icon: Draft_Draft2Sketch
Cannot find icon: Draft_Slope
Cannot find icon: Draft_FlipDimension
Cannot find icon: Draft_2DShapeView
Cannot find icon: Draft_Layer
Cannot find icon: Draft_PlaneProxy
Cannot find icon: Draft_SwitchMode
Cannot find icon: Draft_AddToGroup
Cannot find icon: Draft_SelectGroup
Cannot find icon: Draft_AddConstruction
Cannot find icon: Draft_Heal
Cannot find icon: Draft_Construction
Cannot find icon: Draft_Continue
Cannot find icon: Draft_Apply
Cannot find icon: Draft_SelectPlane
Cannot find icon: Draft_Snap
Cannot find icon: Draft_Grid
Cannot find icon: Draft_AutoGroup
Cannot find icon: Draft_Apply
Cannot find icon: Draft_Arc
Cannot find icon: Draft_BezCurve
Cannot find icon: Draft_Drawing
```

[image]

There are lots of missing icons in the GUI. Notice that Assembly containers have no proper icons too. 

I triple checked that my FreeCAD clone has the right set of commits: 

```
root@fc2:~/FreeCAD# git log --oneline
a491be7f04 (HEAD -> LinkStage3, origin/LinkStage3, origin/HEAD) Sketcher: fix geometry rendering order   
root@fc2:~/FreeCAD# git reset --hard origin/LinkStage3
HEAD is now at a491be7f04 Sketcher: fix geometry rendering order
root@fc2:~/FreeCAD# git remote -v 
origin  https://github.com/realthunder/FreeCAD.git (fetch)
origin  https://github.com/realthunder/FreeCAD.git (push)
```

realthunder, 2020-11-29:
How do you start the FreeCAD, directly run it from the build directory, or from the prefix directory `$FREECAD`? Have you done `make install` after compilation? The error seems to indicate that you have more than one FreeCAD installation, and they clash with each other. BTW, running directly from the build directory is an easy way of having multiple installations. I have several builds on my system for development and testing.

ceremcem, 2020-11-29:
> The error seems to indicate that you have more than one FreeCAD installation,

I was doing `make install` after compilation. I just invoked FreeCAD from the build directory, but the result is the same: 

```
aea@fc2:/root/fc-build/Release/bin$ ./FreeCAD
FreeCAD 0.19, Libs: 0.19R25935 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
During initialization the error "UI file does not exist" occurred in /root/fc-build/Release/Mod/Draft/InitGui.py
Please look into the log file for further information
During initialization the error "UI file does not exist" occurred in /root/fc-build/Release/Mod/Arch/InitGui.py
Please look into the log file for further information
3.994584 <asm3.main> init_gui.py(14): no solver backend found
Cannot find icon: Draft_Dimension
connect failed: No such file or directory
Traceback (most recent call last):
  File "<string>", line 118, in Initialize

UI file does not exist
```

realthunder, 2020-11-30:
Not sure what happen. Did you reuse the same build directory where you previously configured for python2?

ceremcem, 2020-11-30:
No, I also tried to build by removing present build directory. Now I'm removing the `INSTALL_PREFIX:PATH` and such, and strip down the build command like so: 

```
        cmake ../../FreeCAD \
                -DFREECAD_USE_OCC_VARIANT="Official Version" \
                -DCMAKE_BUILD_TYPE=Release \ 
                -DBUILD_QT5=ON \
                -DPYTHON_EXECUTABLE=/usr/bin/python3
```

Then I'll try to run directly from build directory. 

(I've removed the `/opt/FreeCAD` completely just in case)

realthunder, 2020-11-30:
The error seems to indicate something wrong with your PySide2 installation. Maybe you can remove all package related to PySide and reinstall it.

ceremcem, 2020-11-30:
I created a fresh Debian Buster LXC container, followed the official FreeCAD build instructions [as you mentioned](https://github.com/realthunder/FreeCAD_assembly3/issues/350#issuecomment-735029982) and all went okay.

closed 2020-11-30 by ceremcem