realthunder/FreeCAD #78: GUI Esiquises installation and resizing

tiket18, 2020-12-05:
The most terrible situation is when you create a size and the digits of this size get on some line. After that, it's very, very difficult to get to the line with the mouse and move it.

This is a problem common to all assemblies.

It would be right that the lines of diameter sizes should never be created matching the axes. It would be right not to overlay values on lines.
[image]

realthunder, 2020-12-07:
Why not just drag the diameter/length constraint label out of the way.

tiket18, 2020-12-07:
To drag it, you need to be able to select it, and to do this, you need to teach the program to select the size mark, and not the line or axis located there. This is the problem.

tiket18, 2020-12-07:
When the figure lies on top of the line, it is useless to zoom in and poke with the mouse. The axis or line is selected.

realthunder, 2020-12-07:
You can't select the line of a constraint, but you can select the label, the one that's showing the length or diameter.

[image]

realthunder, 2020-12-07:
Maybe you have set the pick radius too big.

[image]

tiket18, 2020-12-07:
я знаю, что выделять надо не линию размера, а цифры. Но если при создании ограничения цифра ложится поверх оси это сделать мягко говоря очень не просто. В этой связи и нужно научить программу не располагать цифры поверх линий.
I know that you need to highlight not the size line, but the numbers. But if, when creating a constraint, the figure lies on top of the axis, this is very difficult to do, to put it mildly. In this regard, it is necessary to teach the program not to place numbers over the lines.

tiket18, 2020-12-07:
Yes, thank you, it helped - not so much the capture radius as the font size. But this is a bad way - with a rich sketch, a large font is very difficult.
[image]
[image]

tiket18, 2020-12-15:
No, not hitting the mark and not resizing. This has to be fixed - it's not nice to spend so much time and effort on catching the size mark.
[size.FCStd.zip](https://github.com/realthunder/FreeCAD/files/5696310/size.FCStd.zip)
[image]
[image]
[image]

realthunder, 2020-12-17:
Can you please capture a screencast video or gif to explaining your problem? I still don't quite get your problem yet.

Shkolik, 2022-07-15:
Since radius/diameter constraints now creates with some angle to the axis by default there in no this issue anymore.

closed 2022-07-15 by tiket18