realthunder/FreeCAD #80: Broke the Path workbench

tiket18, 2020-12-16:
When you open a projectst file in which a Path job was previously created, the Path tree is broken. 
Here is a screenshot, opened the same file in two versions of the program - on the left was so and in this thread last week, on the right - as in the claimed build.

[image]
OS: Arch Linux (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26091 (Git)
Build type: Release
Branch: makepkg
Hash: eb10f4011b36086ed70e8f667c70c10c6d2ecb3e
Python version: 3.8.6
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Russian/Russia (ru_RU)

realthunder, 2020-12-17:
Can you please attach a document here with the broken tree?

tiket18, 2020-12-17:
[sdvd2.FCStd.zip](https://github.com/realthunder/FreeCAD/files/5708020/sdvd2.FCStd.zip)
[Uploading sfer.FCStd.zip…]()

realthunder, 2020-12-17:
Looks like the upstream Path module has broken backward compatibility. I get the same broken tree view when using the latest upstream AppImage 23323.

realthunder, 2020-12-17:
My mistake. It has nothing to do with upstream. It seems the file you attached is saved while FC is having trouble loading PathGui module. Maybe it is related to your Path compilation problem somehow?

tiket18, 2020-12-17:
I have attached two different files. One, the small one, is saved in the FC version, which already has the problem. The other (the big one) was saved earlier, in the locally built version of FC, in which the problem was not yet present. The problem appeared after several fixes in the program code from December 12-12. The left-to-right screenshots were from the large file, which was opened simultaneously in appimage upstream and in the local build, which was built without errors.
The bugs with the build came later and therefore the build was not born with bugs :)

closed 2020-12-19 by tiket18