realthunder/FreeCAD #81: ninja: error: 'Boost::python-NOTFOUND', needed by 'Mod/Path/area.so', missing and no known rule to make it

tiket18, 2020-12-16:
ninja: error: 'Boost::python-NOTFOUND', needed by 'Mod/Path/area.so', missing and no known rule to make it

realthunder, 2020-12-17:
Have you installed boost-python dependency?

tiket18, 2020-12-17:
yaourt boost-python 
1 aur/boost-python2 1.74.0-2 [installed] (0) (0,00)
    Free peer-reviewed portable C++ source libraries - development headers
2 aur/boost-python2-libs 1.74.0-2 [installed] (0) (0,00)
    Free peer-reviewed portable C++ source libraries - runtime libraries
3 aur/mingw-w64-boost-python 1.73.0-1 (0) (0,00)
    Free peer-reviewed portable C++ source libraries (mingw-w64, python)

realthunder, 2020-12-17:
I am not familiar with Arch Linux, but you are probably building with Python 3 right? You'll need boost python for Python3, too. Maybe you can check out the recipe [here](https://aur.archlinux.org/packages/freecad-git/)?

tiket18, 2020-12-17:
Well, it used to work somehow with what we have. But installing something additional will be a problem now - Arch python is already version 3.9 and new packages pull 3.9 dependencies, and FC is not ready for this yet.

tiket18, 2020-12-17:
In my case the Area.cpp.o file is nowhere to be found. 

I will try to roll back the state of the root file system to minus two weeks and give it a try. Probably at night.

closed 2020-12-19 by tiket18