realthunder/FreeCAD #82: Not possible to fillet a PAD created from B-Spline

projectcad, 2020-12-21:
Not possible to fillet a PAD (in Part Design WB ) created from B-Spline

ceremcem, 2020-12-21:
Please provide reproduction steps and version info of FreeCAD you are using, so we may at least confirm the issue with the latest commit if you're not using the latest commit.

projectcad, 2020-12-24:
Tried it in two ones..but same.. Not possible to fillet the PAD created from B - Spline.... (in Part Design WB, 
Body >Sketch >Closed 5-6 pointed B-Spline>PAD> Fillet) 

1-) Freecad 0.19 Linkstage3 branch version Git 25985

2-) Freecad Daily 0.19....2020/12/14 21:36

projectcad, 2021-02-09:
By the way, "chamfer" is possible on this object mentioned above. @realthunder

projectcad, 2021-03-21:
It is possible any more...thank you @realthunder 
[image]

https://github.com/realthunder/FreeCAD/commit/bdbff46007be5f7f53e229ac19ff5db7177fe948

CC : @ceremcem

closed 2021-03-21 by projectcad