realthunder/FreeCAD #84: Unable to set Attachment Offset of substractive primitive in Part Design Workbench

frascuelillo, 2020-12-28:
Unable to set Attachment Offset of any substractive primitive, and if I try to cancel Freecad crash:

[image]

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.08.23076 +2912 (Git)
Build type: Release
Branch: LinkStage3
Hash: 1eba27283118bae43f5264d38caf7451b848f49a
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Spanish/Spain (es_ES)

realthunder, 2020-12-29:
Please try again with the latest release 12.29

frascuelillo, 2020-12-29:
> Please try again with the latest release 12.29

The problem persists, the sustractive primitive preview is always at origin.

realthunder, 2020-12-30:
Please try release 12.30

frascuelillo, 2020-12-30:
Now it has a extrange behaivour, by default there is no preview, wen click on preview only wires is shown, when click Show on top, the preview appears. If not selected Show preview neither Show on top, if I change attachement offset the created object moves.
When cancel, an error appears in Report View widow:

[image]

realthunder, 2020-12-31:
You current choice of preview on/off setting will be remembered and used in the next edit. For the wire only problem, switch to Part design workbench and then go to Preference -> Part design -> Shape appearance, set some alpha value of Additive/Subtractive/Dress up color. I'll apply some transparency clamp for these colors later.

frascuelillo, 2020-12-31:
Certainly the cable problem comes out when setting the alpha value!
I have tried to read and understand the code to try to help, but it far exceeds my limited knowledge of programming and how FreeCad's code is structured.
Let me thank you for the effort you are doing with this branch, the work you are doing is really impressive.

Shkolik, 2022-07-15:
Looks like it's working. @luzpaz maybe it's time to close issue?

closed 2022-07-15 by luzpaz