realthunder/FreeCAD #85: [FEM] Unable to get material's Young's Modulus

YgorSouza, 2020-12-28:
When I create a FEM analysis and pick a material out of the list, the Young's Modulus value stays at zero. It cannot be edited manually, so it is no possible to run the analysis. Moreover, when I confirm the task pane and open it again, the material card is set to "_document_material", and not the one I had chosen. If I open the same file in upstream FreeCAD, I am able to select the same material and run the analysis without any problem.

The following line in the program's output might explain it. it is not present the upstream version:

`YoungsModulus in material data seems to have no unit or a wrong unit (reset the value): 1C22`

[fem_test_asm3.FCStd.zip](https://github.com/realthunder/FreeCAD/files/5748446/fem_test_asm3.FCStd.zip)

OS: Ubuntu 20.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.17.23076 +3024 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 560eb2c4645cd548b6f718c645c7af70384ff54c
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Upstream version tested for comparison:

OS: Ubuntu 20.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23546 (Git) AppImage
Build type: Release
Branch: master
Hash: 6b017f9a16b15b0e628c8d874c4058442dee5548
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

YgorSouza, 2020-12-28:
I don't know if the issue was introduced in this branch, but I couldn't find any recent commit in upstream that would explain why it happens here and not there, so I decided to open the issue here just in case.

realthunder, 2020-12-29:
Thanks for reporting. This is a duplicate of realthunder/FreeCAD_Assembly3#390. It's been fixed already. I'll make a new release soon.

YgorSouza, 2020-12-29:
Oh, I'm sorry I opened a duplicate. I didn't think this was relevant to Assembly 3, so I didn't check there. I'll check next time.

realthunder, 2020-12-29:
No problem. Actually, you are right to report it here, as it is not related to assembly. It's just that other people might not know where to report exactly. I am fine with either place.