realthunder/FreeCAD #86: [UI] DockWindow transparency

obelisk79, 2020-12-31:
I recently switched to linux from windows and have been setting everything up with the latest daily (12/30) to get back to production like I'm used to.

However I seem unable to get truly transparent dockwindows. They have a translucent grey background instead. I hope I'm just missing a setting somewhere. I've taken a screenshot where you can see the translucency how I positioned the navi-cube.

https://imgur.com/YHut3up.png

realthunder, 2020-12-31:
Select the 'Dark-overlay' as the `Overlay style sheet` in Preferences -> General.

obelisk79, 2020-12-31:
Thanks for the quick response. I feel a bit silly now, but that worked great! Truly appreciate all the work you do on FreeCAD.

closed 2020-12-31 by obelisk79