realthunder/FreeCAD #87: [Feature Request] Solver Status in Tree View.

obelisk79, 2021-01-02:
As illustrated below, I'd like to request consideration of adding a flag that allows tree view to indicate solver status for sketches in tree view as illustrated below. The benefit is to allow a simple way to identify sketches which are not fully constrained. I also created a discussion about this on the freecad forums here: https://forum.freecadweb.org/viewtopic.php?f=8&t=53870

https://files.catbox.moe/ieiwy9.png
https://files.catbox.moe/jz1rub.png

obelisk79, 2021-01-02:
Additionally, would it be simple/easy to allow expanding all sub-elements in a body's feature tree to quickly show/hide the status of all sketches?

obelisk79, 2021-01-03:
I'm closing this because I think a solution is coming along with FC master and I don't want to waste your efforts. Thank you for your time.

closed 2021-01-03 by obelisk79

realthunder, 2021-01-09:
> Additionally, would it be simple/easy to allow expanding all sub-elements in a body's feature tree to quickly show/hide the status of all sketches?

Right click the body and select 'Auto group'. This will group all sketches used in this body under one group object. There also one group fro datums, and another one for holding all other non-PartDesign features, like shape binders. Toggle the visibility of the group will show/hide all its children.