realthunder/FreeCAD #89: compile error on macos with latest head

ipatch, 2021-01-10:
is there a known commit that will successfully build on macos?

```shell
cmake \
    -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG\
    -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG\
    -DCMAKE_INSTALL_LIBDIR=lib\
    -DCMAKE_BUILD_TYPE=Release\
    -DCMAKE_FIND_FRAMEWORK=LAST \
    -DCMAKE_VERBOSE_MAKEFILE=ON \
    -Wno-dev \
    -DCMAKE_OSX_SYSROOT="/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk" \
    -std="c++14" \
    -DCMAKE_CXX_STANDARD=14 \
    -DBUILD_ENABLE_CXX_STD:STRING=C++14 \
    -Wno-deprecated-declarations \
    -DBUILD_QT5=ON \
    -DUSE_PYTHON3=1\
    -DPYTHON_EXECUTABLE="/usr/local/bin/python3" \
    -DBUILD_FEM_NETGEN=1 \
    -DBUILD_FEM=1 \
    -DBUILD_TECHDRAW=0 \
    -DFREECAD_USE_EXTERNAL_KDL=ON \
    -DFREECAD_CREATE_MAC_APP=OFF \
    -DCMAKE_PREFIX_PATH="/usr/local/opt/qt/lib/cmake;\
    /usr/local/opt/nglib/Contents/Resources;\
    /usr/local/opt/vtk@8.2/lib/cmake;\
    /usr/local/opt/icu4c/lib;\
    /usr/local;"\
    -DCMAKE_INSTALL_PREFIX="/opt/beta/freecad.real.thunder" ../freecad/
```

```shell
/opt/code/github/public/forks/freecad-git/freecad/src/Gui/moc_DlgPropertyLink.cpp:151:16: error: static_cast from 'Gui::Dialog::DlgPropertyLink *' to 'Gui::SelectionObserver *', which are not related by inheritance, is not allowed
        return static_cast< Gui::SelectionObserver*>(this);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

realthunder, 2021-01-10:
The error does not seem to make any sense. Can you please attach the file `/opt/code/github/public/forks/freecad-git/freecad/src/Gui/moc_DlgPropertyLink.cpp` here?

I just tried on MacOS and does not have any error. It conda based. I'm use the script (here)[https://github.com/realthunder/FreeCADMakeImage].

ipatch, 2021-01-10:
okay, yes i attach file. give a sec. and i'm relying on dependencies installed via brew, that might make a difference.

ipatch, 2021-01-11:
hey doing a little further investigation seems the build will more than likely finish ie. my current build process is 85%, and i believe my issue stems from doing in source builds within the source directory and not within an isolated build directory and due to the `.gitignore` ignoring certain build artifacts it wasn't immediately clear to me that my freecad source directory had quite a bit of build artifacts ie. Makefiles and CMake related files nested through the directory structure, after removing those the build is moving along.  Sorry for the false positive, and thank you for making such an ambitious fork, look forward to using it first hand.

closed 2021-01-11 by ipatch