realthunder/FreeCAD #91: macos build failure with latest commit, commit # 26157

ipatch, 2021-01-11:
hey, please excuse if this is another issue related to my local settings or build directories.

i did a, 

- fresh clone of the _LinkStage3_ branch
- created an empty build directory 
- used the below cmake settings
- then ran `make -j8`

the build fails when reaching the PartDesign of the build process

```shell
opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/Command.cpp:2030:56: error: use of undeclared identifier '_2'
            boost::bind(&commandOverride, this, 0, _1, _2), "Part_Thickness");                                                    ^
```

that is not the error but there are a few more after that with the same error message related to boost

```shell
/opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/Command.cpp:2030:56: error: use of undeclared identifier '_2'
            boost::bind(&commandOverride, this, 0, _1, _2), "Part_Thickness");
                                                       ^
/opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/Command.cpp:2147:52: error: use of undeclared identifier '_1'; did you mean 'boost::placeholders::_1'?
            boost::bind(&commandOverride, this, 0, _1, _2), "Part_Mirror");
                                                   ^~
                                                   boost::placeholders::_1
/usr/local/include/boost/bind/placeholders.hpp:58:38: note: 'boost::placeholders::_1' declared here
BOOST_STATIC_CONSTEXPR boost::arg<1> _1;
                                     ^
/opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/Command.cpp:2147:56: error: use of undeclared identifier '_2'
            boost::bind(&commandOverride, this, 0, _1, _2), "Part_Mirror");
                                                       ^
/opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/Command.cpp:2287:20: warning: lambda capture 'pcActiveBody' is not used [-Wunused-lambda-capture]
    auto worker = [pcActiveBody, cmd](App::DocumentObject *Feat,
                   ^~~~~~~~~~~~~
/opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/Command.cpp:2393:20: warning: lambda capture 'pcActiveBody' is not used [-Wunused-lambda-capture]
    auto worker = [pcActiveBody, cmd](App::DocumentObject *Feat,
                   ^~~~~~~~~~~~~
/opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/Command.cpp:2539:52: error: use of undeclared identifier '_1'; did you mean 'boost::placeholders::_1'?
            boost::bind(&commandOverride, this, 0, _1, _2), "Part_Boolean");
                                                   ^~
                                                   boost::placeholders::_1
/usr/local/include/boost/bind/placeholders.hpp:58:38: note: 'boost::placeholders::_1' declared here
BOOST_STATIC_CONSTEXPR boost::arg<1> _1;
                                     ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
275 warnings and 20 errors generated.
make[2]: *** [src/Mod/PartDesign/Gui/CMakeFiles/PartDesignGui.dir/Command.cpp.o] Error 1
In file included from /opt/code/github/public/forks/freecad-git/freecad-src-realthunder/src/Mod/PartDesign/Gui/CommandBody.cpp:30:
In file included from /usr/local/include/opencascade/TopExp_Explorer.hxx:26:
In file included from /usr/local/include/opencascade/TopoDS_Shape.hxx:24:
/usr/local/include/opencascade/TopLoc_Location.hxx:94:3: warning: use of the 'nodiscard' attribute is a C++17 extension [-Wc++17-extensions]
  Standard_NODISCARD Standard_EXPORT TopLoc_Location Inverted() const;
  
```

at the end of the build process output i see

```
273 warnings generated.
make[1]: *** [src/Mod/PartDesign/Gui/CMakeFiles/PartDesignGui.dir/all] Error 2
make: *** [all] Error 2
```

i'm using boost provided by homebrew

```shell
brew info boost
boost: stable 1.75.0 (bottled), HEAD
Collection of portable C++ source libraries
https://www.boost.org/
/usr/local/Cellar/boost/1.75.0_1 (15,058 files, 601.5MB) *
  Poured from bottle on 2021-01-04 at 21:21:52
From: https://github.com/ipatch/homebrew-core/blob/HEAD/Formula/boost.rb
License: BSL-1.0
==> Dependencies
Required: icu4c ✔
==> Options
--HEAD
        Install HEAD version
```

<details>

<summary>cmake args</summary>

```shell
cmake \                                                                should-You | 0 < 12:58:50
    -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG\
    -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG\
    -DCMAKE_INSTALL_LIBDIR=lib\
    -DCMAKE_FIND_FRAMEWORK=LAST \
    -DCMAKE_VERBOSE_MAKEFILE=ON \
    -Wno-dev \
    -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk \
    -std=c++14 \
    -DCMAKE_CXX_STANDARD=14 \
    -DBUILD_ENABLE_CXX_STD:STRING=C++14 \
    -Wno-deprecated-declarations \
    -DBUILD_QT5=ON \
    -DUSE_PYTHON3=1\
    -DPYTHON_EXECUTABLE="/usr/local/bin/python3" \
    -DBUILD_FEM_NETGEN=1 \
    -DBUILD_FEM=1 \
    -DBUILD_TECHDRAW=0 \
    -DFREECAD_USE_EXTERNAL_KDL=ON \
    -DCMAKE_PREFIX_PATH="/usr/local/opt/qt/lib/cmake;\
    /usr/local/opt/nglib/Contents/Resources;\
    /usr/local/opt/vtk@8.2/lib/cmake;\
    /usr/local/opt/icu4c/lib;\
    /usr/local;"\
    -DCMAKE_BUILD_TYPE=Release\
    -DFREECAD_CREATE_MAC_APP=OFF \
    -DCMAKE_INSTALL_PREFIX="/opt/beta/freecad.real.thunder" ../freecad-src-realthunder
```

</details

ipatch, 2021-01-13:
i far from an active cpp developer, so please forgive me if i'm any of this stuff wrong but i feel somewhat obliged to adding feedback to this issue just for the fact that I opened it.

did some reading this afternoon and came across some relevant links,

---

https://stackoverflow.com/questions/13596697/c11-placeholders-with-boost
https://theboostcpplibraries.com/boost.bind

---

it seems using `_1` and `_2` is part of the cpp anonymous namespace and that using the boost.bind calls seems to trigger compiler errors due to not knowing the namespace to use. so adding the below lines to the offending files

```
using boost::placeholders::_1;
using boost::placeholders::_2;
```

has bumped the build process further along, that is not say there will not be runtime errors, or possibly further compile time errors.

---

basically leaving this here for future me 👴🏼 and spiders 🕷

realthunder, 2021-01-13:
Thanks for the investigation. This is indeed a problem about namespace conflict. I was in the middle of a big merge to sync with the upstream so cannot go back and apply the patch in `LinkStage3`. You can either manually apply the fix in [this](https://github.com/realthunder/FreeCAD/commit/2a768f92ac961f0fbd9cc8f9b8097d45dfa2a001) commit, or try `LinkMerge` branch which is where I am doing the merge. But please be aware that there might some potential problems with sketch as there are a lot of changes in upstream.

closed 2021-01-14 by ipatch