realthunder/FreeCAD #92: Build fails (on Fedora 33)

christian-nils, 2021-01-26:
Hi,

I tried to build your branch LinkStage3, and it fails when compiling the file `moc_PieMenu.cpp`
Error message (https://paste.ubuntu.com/p/vv6QJWV7Pj/)

My cmake summary report is there: https://paste.ubuntu.com/p/p2YQQdTS92/

The issue seems to be coming from the `initMenu()` class member that would be missing from the class PieMenu.

Any ideas where the problem would come from?

I can give more information about my setup if required.

Thanks! :)

realthunder, 2021-01-27:
Not sure if the above commit will fix your problem. Please try.

christian-nils, 2021-01-28:
Yes, it does! Thanks! :)

christian-nils, 2021-02-10:
Hi,

Reopening this issue because I still cannot build it completely LinkStage3 (at commit: 2155aed7cf3c8e6e93ca38b91e6f7914ababa836) 

I have this error message:

`/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayProxyWidget::OverlayProxyWidget(Gui::OverlayTabWidget*)':
OverlayWidgets.cpp:(.text+0x243): undefined reference to `vtable for Gui::OverlayProxyWidget'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0x255): undefined reference to `vtable for Gui::OverlayProxyWidget'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayToolButton::OverlayToolButton(QWidget*)':
OverlayWidgets.cpp:(.text+0xe4d): undefined reference to `vtable for Gui::OverlayToolButton'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0xe5f): undefined reference to `vtable for Gui::OverlayToolButton'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayTabWidget::OverlayTabWidget(QWidget*, Qt::DockWidgetArea)':
OverlayWidgets.cpp:(.text+0xf14): undefined reference to `vtable for Gui::OverlayTabWidget'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0xf29): undefined reference to `vtable for Gui::OverlayTabWidget'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayTitleBar::OverlayTitleBar(QWidget*)':
OverlayWidgets.cpp:(.text+0x90fe): undefined reference to `vtable for Gui::OverlayTitleBar'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0x9110): undefined reference to `vtable for Gui::OverlayTitleBar'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayDragFrame::OverlayDragFrame(QWidget*)':
OverlayWidgets.cpp:(.text+0x9d3a): undefined reference to `vtable for Gui::OverlayDragFrame'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0x9d4c): undefined reference to `vtable for Gui::OverlayDragFrame'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlaySizeGrip::OverlaySizeGrip(QWidget*, bool)':
OverlayWidgets.cpp:(.text+0x9ec7): undefined reference to `vtable for Gui::OverlaySizeGrip'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0x9ed9): undefined reference to `vtable for Gui::OverlaySizeGrip'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlaySizeGrip::mouseMoveEvent(QMouseEvent*)':
OverlayWidgets.cpp:(.text+0xa17b): undefined reference to `Gui::OverlaySizeGrip::dragMove(QPoint const&)'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlaySplitter::OverlaySplitter(QWidget*)':
OverlayWidgets.cpp:(.text+0xa1c8): undefined reference to `vtable for Gui::OverlaySplitter'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0xa1da): undefined reference to `vtable for Gui::OverlaySplitter'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlaySplitterHandle::OverlaySplitterHandle(Qt::Orientation, QSplitter*)':
OverlayWidgets.cpp:(.text+0xa5de): undefined reference to `vtable for Gui::OverlaySplitterHandle'
/usr/bin/ld: OverlayWidgets.cpp:(.text+0xa5f0): undefined reference to `vtable for Gui::OverlaySplitterHandle'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayGraphicsEffect::OverlayGraphicsEffect(QObject*)':
OverlayWidgets.cpp:(.text+0xb5c2): undefined reference to `vtable for Gui::OverlayGraphicsEffect'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayTabWidget::tr(char const*, char const*, int)':
OverlayWidgets.cpp:(.text._ZN3Gui16OverlayTabWidget2trEPKcS2_i[_ZN3Gui16OverlayTabWidget2trEPKcS2_i]+0x2c): undefined reference to `Gui::OverlayTabWidget::staticMetaObject'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayTabWidget* qobject_cast<Gui::OverlayTabWidget*>(QObject*)':
OverlayWidgets.cpp:(.text._Z12qobject_castIPN3Gui16OverlayTabWidgetEET_P7QObject[_Z12qobject_castIPN3Gui16OverlayTabWidgetEET_P7QObject]+0x16): undefined reference to `Gui::OverlayTabWidget::staticMetaObject'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayProxyWidget* qobject_cast<Gui::OverlayProxyWidget*>(QObject*)':
OverlayWidgets.cpp:(.text._Z12qobject_castIPN3Gui18OverlayProxyWidgetEET_P7QObject[_Z12qobject_castIPN3Gui18OverlayProxyWidgetEET_P7QObject]+0x16): undefined reference to `Gui::OverlayProxyWidget::staticMetaObject'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlaySplitterHandle* qobject_cast<Gui::OverlaySplitterHandle*>(QObject*)':
OverlayWidgets.cpp:(.text._Z12qobject_castIPN3Gui21OverlaySplitterHandleEET_P7QObject[_Z12qobject_castIPN3Gui21OverlaySplitterHandleEET_P7QObject]+0x16): undefined reference to `Gui::OverlaySplitterHandle::staticMetaObject'
/usr/bin/ld: CMakeFiles/FreeCADGui.dir/OverlayWidgets.cpp.o: in function `Gui::OverlayTitleBar* qobject_cast<Gui::OverlayTitleBar*>(QObject*)':
OverlayWidgets.cpp:(.text._Z12qobject_castIPN3Gui15OverlayTitleBarEET_P7QObject[_Z12qobject_castIPN3Gui15OverlayTitleBarEET_P7QObject]+0x16): undefined reference to `Gui::OverlayTitleBar::staticMetaObject'
collect2: error: ld returned 1 exit status
make[2]: *** [src/Gui/CMakeFiles/FreeCADGui.dir/build.make:6217: lib/libFreeCADGui.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:5289: src/Gui/CMakeFiles/FreeCADGui.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
`

Do you have any idea why I would get such error? Thanks!

christian-nils, 2021-02-13:
Found the issue, I guess!

When commenting l.115 and 117 of `OverlayWidgets.h` the compilation finishes properly.

```C
// #if QT_VERSION  >= 0x050000
#   define FC_HAS_DOCK_OVERLAY
// #endif
```

@realthunder, can you check?

realthunder, 2021-02-14:
That's strange. What is your Qt version? It seems that somehow you have two conflict macro define of QT_VERSION?

christian-nils, 2021-02-14:
Yes, I had a conflicting issue at the beginning, but now I have fixed it, my config report can be seen here -> https://paste.ubuntu.com/p/8twRgXr9rp/

How can I check the value QT_VERSION?