realthunder/FreeCAD #94: Feature request: Provide a way to set rotation center in Gesture Navigation Style

ceremcem, 2021-01-31:
Setting rotation center is crucial for a proper navigation. Currently there is no gesture to set the rotation center in "Gesture Navigation Style". 

### Proposal 

AFAIK, there is no laptop that can run FreeCAD has 1 button touchpad, so left mouse button's long click event can be used for that purpose.

realthunder, 2021-02-01:
There is actually a key (H) to set rotation center in Gesture Navigation. It will set the rotation center as the last pressed cursor location, either from touchpad, or mouse, or touch screen.

[image]

ceremcem, 2021-02-01:
Ah, yes, there is a way. Sorry, it's embarrassing to miss it. I'm going to close this issue at this point.

closed 2021-02-01 by ceremcem

realthunder, 2021-02-02:
I have added an alternative function of long press which can be selected holding the ALT key. The option in the preference is for selecting the default function, either bring up the context menu, or set rotation center.

[image]

ceremcem, 2021-02-02:
Absolutely great! Closes #95

ceremcem, 2021-07-13:
@realthunder I couldn't figure out how to fire a "mouse up" event after setting the rotation center [on this line](https://github.com/realthunder/FreeCAD/commit/60f2599dea856906f4d59cc091e0713f6049fe2a#diff-76b2d842b8ac050b741529e577356508cb9f8a4af6bb423fa8edc57e9e24fc28R425). It would be a super helpful UX behavior to set the rotation of center *immediately* after the recognition of `long_press`, since this action is done countless times during a session. 

# Current behavior
1. Click a spot for a long duration (no indication of how long)
2. Release
3. Scene is centered to that spot and the very same spot is marked as "rotation center". 

# Desired behavior 
1. Click a spot for a long duration
2. As soon as [`long_press`](https://github.com/realthunder/FreeCAD/commit/60f2599dea856906f4d59cc091e0713f6049fe2a#diff-76b2d842b8ac050b741529e577356508cb9f8a4af6bb423fa8edc57e9e24fc28R425) is detected, scene is centered to that spot and the very same spot is marked as "rotation center".

realthunder, 2021-07-14:
I just added this behavior. However, there is no dedicate event for long pressing. The event won't get trigger by just pressing the button without moving. In other words, you'll need to shake your mouse a bit while pressing to get rotation center set.

BTW, have you tried the 'V, R' command to set rotation center on the center of the currently (pre)selected object?

ceremcem, 2021-07-14:
> BTW, have you tried the 'V, R' command to set rotation center on the center of the currently (pre)selected object?

I didn't know this shortcut, it seems very useful. Shortcuts are generally more powerful than mouse clicks. However, with a different keyboard layout, some shortcuts become nearly unusable because they require both hands to interact (or looking at the keyboard). 

My current version is: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27886 (Git)
Build type: Release
Branch: LinkStage3
Hash: 7437caf7c3e4c6fdcb9f29b6c05596d1268c5248
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

[user.cfg.txt](https://github.com/realthunder/FreeCAD/files/6816555/user.cfg.txt)

Currently I can't feel any difference, is there anything that I miss? What I do to test is: 

1. Locate mouse cursor over an entity. 
2. Start clicking
3. Start shaking mouse cursor
4. Expecting "mouse up" like behavior after `long_click` timeout (after ~1000ms). 
5. Observed: No action.

realthunder, 2021-07-14:
Forgot you are on LinkStage3, the commit is in LinkDaily. Anyway, I have cherry picked to LinkStage3. Please pull and try again.

ceremcem, 2021-07-14:
Ah, yes, I should notice that too. I compiled now and it works very well. Thank you.