realthunder/FreeCAD #96: Error - destruction of the model when one of the elements is removed from the body

tiket18, 2021-02-02:
[image]
[image]

Here is a ready and made without errors body, Sketch016 is made on the surface in it, after its removal the model is destroyed. This is a fairly old bug in the program, it's just that earlier projects were more complex and (probably) it would have been harder to "catch" the error.

I sent the file by e-mail.

realthunder, 2021-02-02:
This is not a bug per sa. But it's handling can be improved. The reason for this is because you are cross linking external geometry in `Sketch015`. The sketch is referring to geometries that are outside of the body. This is not allowed. With recent changes in sketch external geometry selection, this should not be possible. The correct way is to use a (green) SubShapeBinder to import external geometries into the body, and then map external geometry on the binder.

I'll probably change the error handling by output some warning only.

luzpaz, 2022-01-19:
@realthunder good to close?

luzpaz, 2022-07-04:
@realthunder ping

closed 2022-07-04 by tiket18